microsoft/AirSim #2276: could not get the image from the camera when I reshape the size of image

tw200464tw, 2019-10-26:
`def get_image(client):
    image = client.simGetImages([airsim.ImageRequest("fpv",airsim.ImageType.Scene,False,False)])[0]
    imageld = np.fromstring(image.image_data_uint8,dtype=np.uint8)
    print(np.array(imageld).shape)
    image_rgb = imageld.reshape(image.height,image.width,3)
    return image_rgb`

just like this ,there comes the error when I reshape the size of image. 
"  Message=cannot reshape array of size 147456 into shape (144,256,3)
  Source=D:\epic_games_projrcts\airsim_demo_test0\python_project\hello_car_one.py
  StackTrace:
  File "D:\epic_games_projrcts\airsim_demo_test0\python_project\hello_car_one.py", line 16, in get_image
    image_rgb = imageld.reshape(image.height,image.width,3)
  File "D:\epic_games_projrcts\airsim_demo_test0\python_project\hello_car_one.py", line 28, in <module>
    image_rgb = get_image(client)
"
But these codes are what I found from the official example.So what is the problem?

MoritzGa, 2019-11-08:
That's my code snippet:

```python
image_response = client.simGetImages([airsim.ImageRequest(0, airsim.ImageType.Scene, False, False)])[0]
image_1d = np.frombuffer(image_response.image_data_unit8, dtype=np.uint8)
image_rgb = image_1d.reshape(image_response.height, image_response_width, 3)
```

ironclownfish, 2020-03-12:
147456 = 256 * 144 * 4
so you need to change your 3 to a 4.
Apologies if we have an official example with this mistake. If you can point it out I'll get it fixed.

closed 2020-03-12 by ironclownfish