microsoft/AirSim #2280: Build FAILED [Windows] - No such file or directory ...\Community\AirSim\AirLib\AirLib.vcxproj

lloydrayner, 2019-10-28:
Hi, 

I am trying to build on Windows 10 and receive the below error. I am running VS 2017 and have tried in both 'Developer Command Prompt for VS 2017' and 'Select x64 Native Tools Command Prompt for VS 2017'

Any help would be appreciated

```
Build FAILED.

"C:\Program Files\Microsoft Visual Studio\2017\Community\AirSim\AirSim.sln" (default target) (1) ->
"C:\Program Files\Microsoft Visual Studio\2017\Community\AirSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
"C:\Program Files\Microsoft Visual Studio\2017\Community\AirSim\AirLib\AirLib.vcxproj" (default target) (3) ->
(ClCompile target) ->
  c:\program files\microsoft visual studio\2017\community\airsim\airlib\include\common\vectormath.hpp(12): fatal error
C1083: Cannot open include file: 'Eigen/Dense': No such file or directory [C:\Program Files\Microsoft Visual Studio\201
7\Community\AirSim\AirLib\AirLib.vcxproj]
  c:\program files\microsoft visual studio\2017\community\airsim\airlib\include\common\vectormath.hpp(12): fatal error
C1083: Cannot open include file: 'Eigen/Dense': No such file or directory [C:\Program Files\Microsoft Visual Studio\201
7\Community\AirSim\AirLib\AirLib.vcxproj]
  c:\program files\microsoft visual studio\2017\community\airsim\airlib\include\common\vectormath.hpp(12): fatal error
C1083: Cannot open include file: 'Eigen/Dense': No such file or directory [C:\Program Files\Microsoft Visual Studio\201
7\Community\AirSim\AirLib\AirLib.vcxproj]
  c:\program files\microsoft visual studio\2017\community\airsim\airlib\include\common\vectormath.hpp(12): fatal error
C1083: Cannot open include file: 'Eigen/Dense': No such file or directory [C:\Program Files\Microsoft Visual Studio\201
7\Community\AirSim\AirLib\AirLib.vcxproj]
  c:\program files\microsoft visual studio\2017\community\airsim\airlib\include\common\vectormath.hpp(12): fatal error
C1083: Cannot open include file: 'Eigen/Dense': No such file or directory [C:\Program Files\Microsoft Visual Studio\201
7\Community\AirSim\AirLib\AirLib.vcxproj]
  c:\program files\microsoft visual studio\2017\community\airsim\airlib\include\common\vectormath.hpp(12): fatal error
C1083: Cannot open include file: 'Eigen/Dense': No such file or directory [C:\Program Files\Microsoft Visual Studio\201
7\Community\AirSim\AirLib\AirLib.vcxproj]
  c:\program files\microsoft visual studio\2017\community\airsim\airlib\include\common\vectormath.hpp(12): fatal error
C1083: Cannot open include file: 'Eigen/Dense': No such file or directory [C:\Program Files\Microsoft Visual Studio\201
7\Community\AirSim\AirLib\AirLib.vcxproj]
  c:\program files\microsoft visual studio\2017\community\airsim\airlib\include\common\vectormath.hpp(12): fatal error
C1083: Cannot open include file: 'Eigen/Dense': No such file or directory [C:\Program Files\Microsoft Visual Studio\201
7\Community\AirSim\AirLib\AirLib.vcxproj]
  c:\program files\microsoft visual studio\2017\community\airsim\airlib\include\common\vectormath.hpp(12): fatal error
C1083: Cannot open include file: 'Eigen/Dense': No such file or directory [C:\Program Files\Microsoft Visual Studio\201
7\Community\AirSim\AirLib\AirLib.vcxproj]
```

closed 2019-11-09 by lloydrayner

Jan1902, 2019-11-19:
Since you closed your own issue I am assuming you have fixed the issue yourself. As I am running into the same error at the moment, may I ask how you solved the problem, in case you did?

rahulnema0411, 2021-05-16:
Hey i am experiencing the same issue..please help