microsoft/AirSim #2283: RPC error when calling simListSceneObjects in python

a-maumau, 2019-10-30:
# Getting a error when using the simListSceneObjects() among the Neighborhood (v1.2.0 release) environment.

My environments are  
- Ubuntu 18.04  
- UE4 v4.23.1  
  
Running a code
```
import airsim
import time

client = airsim.CarClient()
client.confirmConnection()
client.enableApiControl(True)
car_controls = airsim.CarControls()

data = client.simListSceneObjects(".*")
print(data)
```
  
will get
```
msgpackrpc.error.RPCError: rpclib: server could not find function 'simListSceneObjects' with argument count 1.
```
  
Am I doing something wrong?

cpow-89, 2019-11-15:
I had the same issue. I tried to train a network for image segmentation. However, the segmentation that comes out of the pre-built environments is not that useful. So I searched for a way to correct them to my needs, and so I found simListSceneObjects. 

@a-maumau The problem with simListSceneObjects is that it was added after the environments where build with 1.2.1. If you look at the source code provided with the environments(https://github.com/microsoft/AirSim/archive/v1.2.1.zip) this function is missing. And indeed if you look at the commit history of RpcLibServerBase.cpp you can find that the function has been committed some month after the environments had been released. 

So my hope is that someone will be so kind as to rebuild the environments with the latest airsim version. That would be nice.

a-maumau, 2019-11-25:
@cpow-89 
Thank you for your information!
Yes, I also found about this and the windows build might have the feature. 

I will keep this issue open for looking forward to some kind person will release latest builds