microsoft/AirSim #2285: Synchronous simContinueForTime?

ANogin, 2019-10-30:
The typical use of `simContinueForTime` (including in AirSim's own examples) seems to be to immediately sleep for just a bit longer than the `simContinueForTime` argument (ideally also accounting for `ClockSpeed`/etc). This seems both unnecessary, unnecessarily slow (if the extra sleep is largish), error-prone (if the extra sleep is not enough, if the clock speed is not accounted for, etc). Would it be possible to have a synchronous version of simContinueForTime that would simply block until after the sim is done?

yannbouteiller, 2019-11-03:
You can implement a kinda synchronous version of simContinueForTime() by doing this: https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/17#issuecomment-549165603

madratman, 2019-11-04:
Hey @ANogin, can you give https://github.com/microsoft/AirSim/pull/2292 a shot?
simPause should now actually pause everything. I'm leaning towards removing simContinueForTime() and let the user just have simPause(True) and simPause(False) (or well simPause() and simUnpause(). 
cc @sytelus

ANogin, 2019-11-05:
> Hey @ANogin, can you give #2292 a shot?
> simPause should now actually pause everything. I'm leaning towards removing simContinueForTime() and let the user just have simPause(True) and simPause(False) (or well simPause() and simUnpause().
> cc @sytelus

I do not see how that would help at all - what I want is to run the simulator for a specified amount of **simulated** time, and then have it pause and tell me when it is done.

madratman, 2019-11-07:
@Anogin, gotcha. I updated the API to do that. Can you give it a shot now? Also, see discussion here https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/113#issuecomment-550451099

yannbouteiller, 2019-11-14:
> @ANogin, gotcha. I updated the API to do that. Can you give it a shot now? Also, see discussion here [microsoft/AirSim-NeurIPS2019-Drone-Racing#113 (comment)](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/113#issuecomment-550451099)

Sorry @madratman  but I didn't see this comment, do you mean that SimContinueForTime() has been updated from using the CPU time to using the simulated time ?

yannbouteiller, 2019-11-14:
(Oh my bad you only made it synchronous, right?)