microsoft/AirSim #2288: build.cmd running into problems..

potBagMeat, 2019-10-31:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

closed 2019-11-01 by potBagMeat

ArielLopes888, 2023-01-31:
Olá! Qual foi a solução?

dianzizs, 2023-11-22:
Hello, can you share the way that you address this problem? Thanks a lot