microsoft/AirSim #2289: Run Error while using DQNcar.py

Ahill743, 2019-10-31:
Hi, 

I have finally got to the stage where I can run the DQNcar.py and once it connects to server the car just constently respawns practically everytime it hits the ground. What is the issue? I have had a look into the script but I am struggling to find any answers there or elsewhere online.
Pic shows the issue, I do have the same issue on City Environment but cannot get a client server connection on AirSimNH for some reason.

Phython 3.5 (cannot get cntk on python 3.7?)
AirSim-v.1.2.2

[image]

------------------------------------ (Remove above before filing the issue) ------------------------------------

zimmy87, 2021-01-07:
possibly related to #1009

zimmy87, 2021-01-15:
This is a separate issue from #1009. I am seeing the same behavior with the new \AirSim\PythonClient\reinforcement_learning\dqn_car.py script in the blocks environment. Additional investigation is required here.

zimmy87, 2021-01-15:
After investigating this further, I found that the dqn_car.py script is designed to only work with the AirSimNH/Neighborhood environment. Please see [here](https://microsoft.github.io/AirSim/reinforcement_learning/#rl-with-car) for additional details.

closed 2021-01-15 by zimmy87