microsoft/AirSim #2301: unspecified iostream_category error

Amigoshan, 2019-11-07:
It might be related to https://github.com/microsoft/AirSim/issues/768 and https://github.com/microsoft/AirSim/issues/735. 
I'm running a packaged environment on the same machine that does the packaging, using Ubuntu 18.04. The environment in Unreal Engine works perfectly, but the packaged one crashes upon opening. The error message is like this:
```
terminating with uncaught exception of type std::__1::ios_base::failure: mkdir failed for path /home/xxxxxxxx/Binaries/LinuxNoEditor/scifi1/Binaries/Linux/scifi1sionTags//UpdateOverlaps///UpdateOverlaps// with errorno 2 and message No such file or directory: unspecified iostream_category error
```
I also looked at the Diagnostics.txt (I omit all the directory info with 'xxx'): 
```
Generating report for minidump

Application version 4.18.3.0
 ... built from changelist 0

OS version Linux 5.0.0-32-generic (network name: xxx)
Running 4 x86_64 processors (8 logical cores)
Exception was "SIGABRT: abort() called"

<SOURCE START>
<SOURCE END>

<CALLSTACK START>
/xxxxxx/Binaries/Linux/scifi1!FLinuxPlatformStackWalk::CaptureStackBackTrace(unsigned long long*, unsigned int, void*) + some bytes [/xxx/UnrealEngine/Engine/Source/Runtime/Core/Private/Linux/LinuxPlatformStackWalk.cpp:1040]
/xxxxxx/Binaries/Linux/scifi1!FGenericPlatformStackWalk::StackWalkAndDump(char*, unsigned long, int, void*) + some bytes [/xxx/UnrealEngine/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformStackWalk.cpp:173]
/xxxxxx/Binaries/Linux/scifi1!FLinuxCrashContext::CaptureStackTrace() + some bytes [Runtime/Core/Public/Containers/ContainerAllocationPolicies.h:332]
/xxxxxx/Binaries/Linux/scifi1!CommonLinuxCrashHandler(FGenericCrashContext const&) + some bytes [/xxx/UnrealEngine/Engine/Source/Runtime/Linux/LinuxCommonStartup/Private/LinuxCommonStartup.cpp:37]
/xxxxxx/Binaries/Linux/scifi1!PlatformCrashHandler(int, siginfo_t*, void*) + some bytes [/xxx/UnrealEngine/Engine/Source/Runtime/Core/Private/Linux/LinuxPlatformCrashContext.cpp:597]
Unknown!/lib/x86_64-linux-gnu/libpthread.so.0(+0x12890) [0x7eff61603890] + some bytes
Unknown!/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7) [0x7eff6027ae97] + some bytes
Unknown!/lib/x86_64-linux-gnu/libc.so.6(abort+0x141) [0x7eff6027c801] + some bytes
Unknown!/xxxxxx/Binaries/Linux/scifi1() [0x670274d] + some bytes
Unknown!/xxxxxx/Binaries/Linux/scifi1() [0x6702841] + some bytes
Unknown!/xxxxxx/Binaries/Linux/scifi1() [0x67025f3] + some bytes
Unknown!/xxxxxx/Binaries/Linux/scifi1() [0x6702076] + some bytes
/xxxxxx/Binaries/Linux/scifi1!common_utils::FileSystem::createDirectory(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + some bytes [/xxx/Plugins/AirSim/Source/AirLib/src/common/common_utils/FileSystem.cpp:65]
/xxxxxx/Binaries/Linux/scifi1!common_utils::FileSystem::getExecutableFolder() + some bytes [/home/xxxxxx/Plugins/AirSim/Source/AirLib/include/common/common_utils/FileSystem.hpp:61]
/xxxxxx/Binaries/Linux/scifi1!ASimHUD::getSettingsText(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) + some bytes [/xxx/Plugins/AirSim/Source/AirLib/include/common/Settings.hpp:53]
/xxxxxx/Binaries/Linux/scifi1!ASimHUD::initializeSettings() + some bytes [/xxx/Plugins/AirSim/Source/SimHUD/SimHUD.cpp:249]
/xxxxxx/Binaries/Linux/scifi1!ASimHUD::BeginPlay() + some bytes [/xxx/Plugins/AirSim/Source/SimHUD/SimHUD.cpp:27]
/xxxxxx/Binaries/Linux/scifi1!AActor::DispatchBeginPlay() + some bytes [Runtime/Core/Public/Containers/UnrealString.h:56]
/xxxxxx/Binaries/Linux/scifi1!AWorldSettings::NotifyBeginPlay() + some bytes [Runtime/Core/Public/UObject/NameTypes.h:1123]
/xxxxxx/Binaries/Linux/scifi1!AGameStateBase::HandleBeginPlay() + some bytes [/xxx/UnrealEngine/Engine/Source/Runtime/Engine/Private/GameStateBase.cpp:177]
/xxxxxx/Binaries/Linux/scifi1!Unknown + some bytes [Runtime/Engine/Classes/Engine/World.h:1880]
/xxxxxx/Binaries/Linux/scifi1!UEngine::LoadMap(FWorldContext&, FURL, UPendingNetGame*, FString&) + some bytes [/xxx/UnrealEngine/Engine/Source/Runtime/Engine/Private/UnrealEngine.cpp:10681]
/xxxxxx/Binaries/Linux/scifi1!UEngine::Browse(FWorldContext&, FURL, FString&) + some bytes [/xxx/UnrealEngine/Engine/Source/Runtime/Engine/Private/UnrealEngine.cpp:9948]
/xxxxxx/Binaries/Linux/scifi1!UGameInstance::StartGameInstance() + some bytes [/xxx/UnrealEngine/Engine/Source/Runtime/Engine/Private/GameInstance.cpp:434]
/xxxxxx/Binaries/Linux/scifi1!UGameEngine::Start() + some bytes [/xxx/UnrealEngine/Engine/Source/Runtime/Engine/Private/GameEngine.cpp:713]
/xxxxxx/Binaries/Linux/scifi1!FEngineLoop::Init() + some bytes [/xxx/UnrealEngine/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:2726]
/xxxxxx/Binaries/Linux/scifi1!GuardedMain(wchar_t const*) + some bytes [/xxx/UnrealEngine/Engine/Source/Runtime/Launch/Private/Launch.cpp:52]
/xxxxxx/Binaries/Linux/scifi1!CommonLinuxMain(int, char**, int (*)(wchar_t const*)) + some bytes [/xxx/UnrealEngine/Engine/Source/Runtime/Linux/LinuxCommonStartup/Private/LinuxCommonStartup.cpp:236]
Unknown!/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7eff6025db97] + some bytes
Unknown!/xxxxxx/Binaries/Linux/scifi1(_start+0x2a) [0x234429a] + some bytes

<CALLSTACK END>

0 loaded modules

Report end!
```

ibrahim0v, 2019-12-16:
I have the same issue can someone help pls

saihv, 2020-01-20:
Which environment is the one that's failing?

closed 2020-08-17 by jonyMarino