microsoft/AirSim #2308: How to display the video effects of three drones at the same time in UE4

heyz1234, 2019-11-10:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

heyz1234, 2019-11-10:
How to display the video effects of three drones at the same time in UE4

saihv, 2020-01-20:
Can you tell us what you mean by video effects? Do you want to visualize the onboard camera views of 3 drones simultaneously?

jonyMarino, 2020-08-28:
Closed due to age and incompleteness

closed 2020-08-28 by jonyMarino