microsoft/AirSim #2311: Focus distance of drones

ppueyor, 2019-11-13:
Hi!

I would like to know if there is a way to change the manual focus distance of the cameras of the drones, either through settings.md or by calling a function, like it is done in UE4 with a Cinematics Camera:

[image]

Is it possible to place a Cinematics Camera on the drones?

surongkai, 2019-11-21:
Hello, I want to know the photo shown in the picture above, where is the focus setting from?

ppueyor, 2019-11-21:
In Unreal Editor 4, if you place a Cinematic Camera.

surongkai, 2019-11-21:
When I perform the HIL simulation in the UE4, after setting the Setting file, I can generate the depth image. The camera used is the camera in the model. In this case, where can the internal reference such as the camera focal length be found? 
thanks！

ppueyor, 2019-11-21:
Yeah, that was my first question.

I can modiffy those values by placing a cinematics camera in the editor of the map, but can't do it through the AirSim API or the Settings file. Seems like the camera integrated in AirSim is a regular camera. I guess it is not possible to place a cinematic camera in order to change those parameters.

Would be interesting though!

surongkai, 2019-11-21:
Is the camera that generates the depth map in airsim the same as the focal length of the Cinematics Camera? If not, if I want to see the internal parameters of the depth camera, where should you find it?
thanks！

ppueyor, 2019-11-25:
The thing is that Unreal API does provice tools to play with the "UCineCameraComponent", where it is possible to change the FocusDistance programatically from their API.

Would be great if AirSim would let us place a CineCameraComponent instead of a simple camera on the drones and change its params through the API.

gudong321, 2022-04-06:
CinemAirSim is just for Linux, but for win?

ppueyor, 2022-04-06:
Hi @gudong321, 

Theoretically, it is also available for Windows, but I didn't have the chance to test it there.

jonyMarino, 2022-05-30:
@gudong321 the cinematographic camera works on windows too!

closed 2022-05-30 by jonyMarino