microsoft/AirSim #2314: how to get GPS data of the car

Nianan0505, 2019-11-14:
Hello.
I want to get the GPS information of the car, but i can't find the api. 
I have got the GPS data of the drone, but it is difficult for me to find it of car
If you tell me this, I will be very very grateful!

doublestrong, 2020-05-04:
maybe try the GPS api (getGpsData) to read that data from a GPS? i didn't find something like gps_position in the CarState constructor.

gurkanctn, 2020-05-23:
For the HelloCar.py application the getGpsData does not seem to work.

Here's the line that calls getGpsData():
`gps_data = client.getGpsData()`

Error:
`AttributeError: 'CarClient' object has no attribute 'getGpsData'`

doublestrong, 2020-05-30:
Hi,

I haven't try the python code. I am using c++ and this is my code getting Gps data:
`auto gps_data = airsim_client_.getGpsData(vehicle_gps_pair.second, vehicle_gps_pair.first);`
Did you add Gps sensor in your .json setting file?

jonyMarino, 2020-08-10:
@gurkanctn I will close this issue. If you have added gps sensor on your .json setting file you can post a bug issue.

closed 2020-08-11 by Nianan0505