microsoft/AirSim #2323: SITL failure with PX4 in ubuntu

xlong0513, 2019-11-22:
Hi, Ubuntu here. i use launch/px4.launch to start  SITL, but it cannot connect to the AirSim. It keeps waiting for the port as the following image shows:
[image]

 And the QGroundControl cannot be connected either.
[image]

But when i launch the "launch/posix_sitl.launch" with gazebo, it works well.

What's the problem?

Sho412, 2019-11-26:
How the version of Px4 is? v1.9.x?

xlong0513, 2019-11-27:
@Sho412   Checkout to 1.8.2, it;s ok now. But how to connect it with ros or mavros? Do u know that? Appreciate it.

Sho412, 2019-11-27:
Sorry I don't know. Also interested in your question.
P.S. If you'd like to use v 1.9, see #2318

xlong0513, 2019-11-27:
OK! Thanks for the reply.

marufino, 2020-01-31:
@xlong0513 you should be able to run px4.launch from the default mavros ros package, with the `fcu_url` changed to `<arg name="fcu_url" default="udp://:14540@127.0.0.1:14557" />`

closed 2020-09-04 by jonyMarino