microsoft/AirSim #2330: Programmatic Vehicle / Camera Control and State Recording in Computer Vision Mode (C++)

changh95, 2019-12-02:
Hi, I am using AirSim to acquire a dataset consisting of RGB images and ground-truth camera poses for developing SLAM applications. I am quite new here - I have just successfully loaded computer vision mode on a custom environment yesterday. 

I wish to design a camera trajectory, such that the camera will move exactly as how I programmed it to move without any external control. This should work in repeatable manner, so I can reproduce the same movement in the following experiment. But currently I am facing the issues below:

- I am not sure where to look at, to programmatically control camera pose in computer vision mode for C++. The tutorial primarily focuses on cars and multirotor. Any API references or code samples will be greatly appreciated.
- Is there any way to change camera parameters (e.g. aperture, ISO, exposure, shutter speed)? On the AirSim tutorial there seems to be options to change FOV and motion blur etc, but on Unreal Engine these parameters can also be changed, so I guess it may be possible to change these on AirSim too.
- In the future, I may need to integrate IMU data as well for visual-inertial SLAM. Is it possible to get IMU data for computer vision mode, or do I need to switch to multirotor?

Thanks a lot,

changh95, 2019-12-10:
Update:

I've achieved programmatic control of camera pose by using simSetVehiclePose().
Also, the camera parameters that I looked for were for motion blur, and on Image API there is an option to fiddle with motion blur.

One more questions left to be solved:

- Is there way to integrate IMU to Computer Vision Mode?

saihv, 2020-01-20:
What kind of data are you exactly looking for in computer vision mode? There are inherently no dynamics in that mode at all, which is why using an IMU becomes tricky.

changh95, 2020-01-20:
Hi @saihv , 

Currently I am building a dataset for visual-inertial navigation whilst the camera looking onto a fiducial marker on the ground. Ideally I am looking to get images, global camera position, global camera orientation, GT linear acceleration, GT angular velocity, IMU linear acceleration, IMU angular velocity, and timestamp.

Since the time I raised the issue, I realised there is no way to generate a smooth trajectory via AirSim API on computer vision mode. On CV mode, the only way to move the camera is by client.simSetVehiclePose(), which basically teleports the camera to a position, which does not exert any force on IMU therefore the added IMU sensor won't read anything. My latest attempt on using IMU on CV mode is referenced on #2381.

changh95, 2020-01-20:
I also found it very cumbersome to generate a smooth trajectory for the camera. Since this is not possible on AirSim, I had to generate the trajectory on native unreal using cinematic sequence, but there seems to be a huge functional mismatch between unreal and AirSim.

I would like to suggest for a feature that we can call API functions to move the camera smoothly on Computer Vision mode, just like how multirotors can.

saihv, 2020-01-21:
> On CV mode, the only way to move the camera is by client.simSetVehiclePose(), which basically teleports the camera to a position, which does not exert any force on IMU therefore the added IMU sensor won't read anything.

You've sort of answered your own question here: is there any reason you're focused on CV mode? Why not try to get the IMU data from multirotor mode? 

> I would like to suggest for a feature that we can call API functions to move the camera smoothly on Computer Vision mode, just like how multirotors can.

What is the 'step size' for positioning you're using when doing multiple simSetVehiclePose() calls? I'd imagine by using a tiny step size, you can essentially eliminate the jerk between positions.

changh95, 2020-01-22:
@saihv, thanks for the reply!

> You've sort of answered your own question here: is there any reason you're focused on CV mode? Why not try to get the IMU data from multirotor mode?

I am trying to reproduce the movement of handheld imaging device with IMU attached onto it (essentially a smartphone). The type of movement I want to achieve is not easy to reproduce in multimotor mode. 

For example, on CV mode it is super easy to change the angle in x direction for 90 degrees, while keeping the angle in y,z direction and xyz positions. It's simply rotating by changing the pitch angle by 90 degrees, changing the viewpoint from front-side to bottom-side. 

This movement cannot be reproduced on multimotor mode - it is not possible to only change the x angle while keeping xyz positions. The multimotor cannot be in air when its propulsions are in horizontal directions... This is why I tried to implement the IMU sensor on CV mode, as I don't get any restrictions in movements.

> What is the 'step size' for positioning you're using when doing multiple simSetVehiclePose() calls? I'd imagine by using a tiny step size, you can essentially eliminate the jerk between positions.

The jerk isn't a big issue here - rather, I would like to discuss about the easiness of programming using the API. On multirotor mode, if I want an incremental movement of the multirotor from the current position to pose A, I just need to use `moveToPositionAsync()` or something like that. But on CV mode, if I want an incremental movement of the camera from the current position to pose A, I need to break down the path into the step sizes, and have to calculate all the pose data for each step, and then make a loop that loads the pose data and performs `simSetVehiclePose()`. If we can have a similar API function that can easily move the camera actor from one pose to another incrementally with a single API function on CV mode, it would be greatly appreciated!

jonyMarino, 2021-01-14:
Hi. Try using this branch:
https://github.com/jonyMarino/AirSim/tree/external_dynamics_simulator_with_example

closed 2021-01-14 by jonyMarino