microsoft/AirSim #2331: Eigen dependency download broken

petergerten, 2019-12-04:
Downloading the Eigen dependency (in setup.sh) does not work anymore.

`wget http://bitbucket.org/eigen/eigen/get/3.3.2.zip`

results in

```
HTTP request sent, awaiting response... 401 Unauthorized
Username/Password Authentication Failed.
```

petergerten, 2019-12-04:
The github mirror works:
https://github.com/eigenteam/eigen-git-mirror/archive/3.3.2.zip

saihv, 2019-12-04:
Works for me on Ubuntu 18.04. Perhaps something was up with your network temporarily.. could you try again?

closed 2019-12-04 by saihv

asherikov, 2019-12-05:
Eigen is migrating to GitLab, they have already updated download links on their website. It is not clear how long the downloads will be available from bitbucket. github mirror is scheduled for removal as well. 
Please update your scripts.

saihv, 2019-12-05:
@asherikov Good catch, thanks for letting us know. I'll update the URLs to gitlab