microsoft/AirSim #2339: Mismatch between simGetVehiclePose() and simSetVehiclePose()

changh95, 2019-12-10:
Currently using the latest version of airsim, on computer vision mode.

I am trying to reproduce the motion from one scene to another. I have recorded pose data over time on one scene onto a csv file by using `simGetVehiclePose()` , and then on the other scene I set the object pose from the pose data I got on the csv file by using `simSetVehiclePose()`. The objects from two scenes have the same starting point and same starting orientation.

What I expected was the exact same motion being reproduced at the new scene. The motion is delievered the same way, but the orientation of the object is wrong. Assumingly, `simSetVehiclePose` should set the pose exactly the same, but it's not working this way. What I've noticed is that the position is delivered correctly, just not the orientation.

What can be causing this?

-----
When I record pose on csv, this is the code I used

data[0].camera_position[0] << "," << data[0].camera_position[1] << "," << data[0].camera_position[2] << "," << data[0].camera_orientation.coeffs()[0] << "," << data[0].camera_orientation.coeffs()[1] << "," << data[0].camera_orientation.coeffs()[2] << "," << data[0].camera_orientation.coeffs()[3]

and when I set the pose, this is the code I used

tx = std::stof(row[0]);
ty = std::stof(row[1]);
tz = std::stof(row[2]);
qw = std::stof(row[3]);
qx = std::stof(row[4]);
qy = std::stof(row[5]);
qz = std::stof(row[6]);
            
Eigen::Vector3f position(tx, ty, tz);
Eigen::Quaternion<float, Eigen::DontAlign> quaternion(qw, qx, qy, qz);
cvPose = Pose(position, quaternion);
client.simSetVehiclePose(cvPose, true);

changh95, 2019-12-10:
Found what's wrong.

I've been getting the quaternion values by calling from response directly - `response.camera_orientation.coeffs`, and I've been putting these values directly into `Eigen::Quaternion<float, Eigen::DontAlign> quaternion(qw, qx, qy, qz)`.

But turns out, when I make quaternion objects, I need to put it in order of 'qx, qy, qz, qw'.

This can be confusing to many, since the API gives quaternion out as 'qw, qx, qy, qz' and the official Eigen library also uses 'qw, qx, qy, qz', but when we make quaternion object we must use 'qx, qy, qz, qw'.

tharindurmt, 2020-11-03:
Although bit late, I can confirm that this issue is no longer there in the master. Just tested it now and `qw, qx, qy, qz` is the order used in creating a  quaternion.

changh95, 2020-11-04:
> Although bit late, I can confirm that this issue is no longer there in the master. Just tested it now and `qw, qx, qy, qz` is the order used in creating a quaternion.

Thanks for the update - I haven't been able to catch up with this bug.
I'm glad it's solved!

closed 2020-11-04 by changh95