microsoft/AirSim #2345: Creating a new vehicle with mass bigger than 3kg

sebbom, 2019-12-12:
Hello AirSim community,

I am in the process of integrating new vehicles in AirSim but I am wondering if it is actually even possible to have heavier platform.
As stated in: [https://github.com/microsoft/AirSim/blob/0c8dc72ed293ac3fc9445bbc41cec3044fdb523b/AirLib/include/vehicles/multirotor/firmwares/simple_flight/SimpleFlightQuadXParams.hpp#L42 ](url)
it seems that there is a limit in a potential mass, is it only due to the specific simpleflight flight control or because of the physics engine?

I have been trying without success to integrate other flight controller tuned for heavier platform without any success and start to consider that the physics engine is specifically tuned for lighter ones.

Thanks in advance,

marcociara379, 2019-12-12:
I can confirm the bug, and I think it relates to [my issue #2341](https://github.com/microsoft/AirSim/issues/2341).

ibrahim0v, 2020-01-16:
@sebbom Hey,
Can you give more details about what you are trying to do? I have integrated my 8kg 35+Nm torque drone with great physical reactions (well, except the collisions and elasticity and stuff like that) but it is working "okay" now

closed 2020-08-17 by jonyMarino