microsoft/AirSim #2348: UE4 4.24 can not support Airsim,how to solve

obitodaitu, 2019-12-17:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

rajat2004, 2019-12-17:
@obitodaitu You should fill the issue format, as in steps already tried, errors, etc.

Regarding Unreal Engine 4.24, have you tried building the AirSim code? If yes, then please add compiler compatibility problems, error messages, etc.
There's an open PR for some depreciation for 4.22 - https://github.com/microsoft/AirSim/pull/1994, will be helpful

obitodaitu, 2019-12-17:
@rajat2004 It‘s working  in UE4 4.23.
Error     MSB3075	The command“E:\UE4\UE_4.24\Engine\Build\BatchFiles\Build.bat DroneEditor Win64 DebugGame -Project="E:\UE\Drone\Drone.uproject" -WaitMutex -FromMsBuild”exited with code 5。Please verify that you have sufficient rights to run this command.	Drone	C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets	44

obitodaitu, 2019-12-18:
I have solved this problem，in CameraDirector.h file,enum class ECameraDirectorMode must be 0 entry.

F14799a, 2020-01-28:
I am having the same problem trying to use AirSim with UE4.24.  I originally built AirSim with 4.18 and was able to successfully follow the tutorial in "Creating custom environment"
Then I downloaded UE4.24 and the same LandscapeMountains.uproject for 4.24 and tried to follow the same steps as for 4.18.  
The first time generated the error list shown in build error log.jpg
[image]

Subsequent attempts generate the output text below without the library errors.
#################
1>------ Build started: Project: LandscapeMountains, Configuration: DebugGame_Editor x64 ------
1>Parsing headers for LandscapeMountainsEditor
1>  Running UnrealHeaderTool "F:\UE24\LandscapeMountains\LandscapeMountains.uproject" "F:\UE24\LandscapeMountains\Intermediate\Build\Win64\LandscapeMountainsEditor\DebugGame\LandscapeMountainsEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
1>F:/UE24/LandscapeMountains/Plugins/AirSim/Source/CameraDirector.h(24) : LogCompile: Error: 'ECameraDirectorMode' does not have a 0 entry! (This is a problem when the enum is initalized by default)
1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets(44,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.24\Engine\Build\BatchFiles\Build.bat" LandscapeMountainsEditor Win64 DebugGame -Project="F:\UE24\LandscapeMountains\LandscapeMountains.uproject" -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have sufficient rights to run this command.
1>Done building project "LandscapeMountains.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
###############

I see the lack of a 0 entry referenced by obitodaitu above and incrimented all classes under 
ECameraDirectorMode by -1 so there was now 0-7 instead of 1-8

[image]

Then I just incremented the first class from 1 to 0 while keeping 2-8
[image]

Would appreciate steps to troubleshoot.  I have not rebuilt AirSim after downloading 4.24 as I did not see anything specifying the version when building.

pgashwin, 2020-01-29:
Tried the same solution that @obitodaitu suggested. And it worked on UE4.24.1

F14799a, 2020-01-29:
@pgashwin ,
Can you confirm how you implemented the fix?  In Cameradirector.h, I assume the applicable code is:

##########
UENUM(BlueprintType)
enum class ECameraDirectorMode : uint8
{
    CAMERA_DIRECTOR_MODE_FPV = 1	UMETA(DisplayName = "FPV"),
    CAMERA_DIRECTOR_MODE_GROUND_OBSERVER = 2	UMETA(DisplayName = "GroundObserver"),
    CAMERA_DIRECTOR_MODE_FLY_WITH_ME = 3	UMETA(DisplayName = "FlyWithMe"),
    CAMERA_DIRECTOR_MODE_MANUAL = 4	UMETA(DisplayName = "Manual"),
    CAMERA_DIRECTOR_MODE_SPRINGARM_CHASE = 5	UMETA(DisplayName = "SpringArmChase"),
    CAMERA_DIRECTOR_MODE_BACKUP = 6     UMETA(DisplayName = "Backup"),
    CAMERA_DIRECTOR_MODE_NODISPLAY = 7      UMETA(DisplayName = "No Display"),
	CAMERA_DIRECTOR_MODE_FRONT = 8	UMETA(DisplayName = "Front")
};
##########

Did you add a 0 class and make uint9 or renumber them as below and keep uint8?

##########
UENUM(BlueprintType)
enum class ECameraDirectorMode : uint8
{
    CAMERA_DIRECTOR_MODE_FPV = 0	UMETA(DisplayName = "FPV"),
    CAMERA_DIRECTOR_MODE_GROUND_OBSERVER = 1	UMETA(DisplayName = "GroundObserver"),
    CAMERA_DIRECTOR_MODE_FLY_WITH_ME = 2	UMETA(DisplayName = "FlyWithMe"),
    CAMERA_DIRECTOR_MODE_MANUAL = 3	UMETA(DisplayName = "Manual"),
    CAMERA_DIRECTOR_MODE_SPRINGARM_CHASE = 4	UMETA(DisplayName = "SpringArmChase"),
    CAMERA_DIRECTOR_MODE_BACKUP = 5     UMETA(DisplayName = "Backup"),
    CAMERA_DIRECTOR_MODE_NODISPLAY = 6      UMETA(DisplayName = "No Display"),
	CAMERA_DIRECTOR_MODE_FRONT = 7	UMETA(DisplayName = "Front")
};
##########
I apologize if I am asking a very basic question.

pgashwin, 2020-01-30:
@F14799a , I renumbered them to begin from 0 and it worked. However, when you fly the copter, it doesn't default to spring arm chase, but you can always change that with a key-press.

LSBOSS, 2020-01-31:
@pgashwin In my pull request #2402 I also did the renumbering thing but also took care that default camera views per vehicle are the same as before.

Additional, I did some #include fixes because it seems that UnrealEngine 4.24 changed some include paths. Nothing fancy though :)

F14799a, 2020-02-14:
@LSBOSS 

Thank you for the changes in #2402, I implemented those changes in my code, but I also had to change the following to get it working on my PC.

Next error was a missing 'PreWindowsApi.h' in F:\UE24\LandscapeMountains\Plugins\AirSim\Source\AirLib\include\common\common_utils\WindowsApisCommonPre.hpp
#include "PreWindowsApi.h"
#include "AllowWindowsPlatformTypes.h"
#include "AllowWindowsPlatformAtomics.h"
Error fixed by editing to:
#include "windows/PreWindowsApi.h"
#include "windows/AllowWindowsPlatformTypes.h"
#include "windows/AllowWindowsPlatformAtomics.h"

Next error was a missing 'HideWindowsPlatformAtomics.h' in F:\UE24\LandscapeMountains\Plugins\AirSim\Source\AirLib\include\common\common_utils\WindowsApisCommonPost.hpp
#include "HideWindowsPlatformAtomics.h"
#include "HideWindowsPlatformTypes.h"
#include "PostWindowsApi.h"
Error fixed by editing to:
#include "windows/HideWindowsPlatformAtomics.h"
#include "windows/HideWindowsPlatformTypes.h"
#include "windows/PostWindowsApi.h"

This was a warning more than an error:
1>F:/UE24/LandscapeMountains/Plugins/AirSim/Source/RenderRequest.cpp(83): warning C4996: Please use ENQUEUE_RENDER_COMMAND instead Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
1>F:/UE24/LandscapeMountains/Plugins/AirSim/Source/SimHUD/SimHUD.cpp(2): fatal error C1083: Cannot open include file: 'ConstructorHelpers.h': No such file or directory

	####This generated an error:
	ENQUEUE_UNIQUE_RENDER_COMMAND_ONEPARAMETER(
	        SceneDrawCompletion,
	        RenderRequest *, This, this,
	        {
	                This->ExecuteTask();
	        }
	);
	
	####was fixed by:
	RenderRequest* This = this;
	
	ENQUEUE_RENDER_COMMAND(SceneDrawCompletion)(
	[This](FRHICommandListImmediate& RHICmdList)
	{
	    This->ExecuteTask();
	});

##Reference Link: https://forums.unrealengine.com/development-discussion/engine-source-github/1608365-how-to-use-enqueue_render_command-instead-of-enqueue_unique_render_command_oneparameter

closed 2020-03-23 by ironclownfish

koosad, 2021-01-26:
UE_LOG(LogUWPMoviePlayer, Log, TEXT("Starting playback of %s"), MediaItem->Source->Uri->DisplayUri->Data());

		Windows::Media::MediaProperties::VideoEncodingProperties^ VideoProperties = MediaItem->VideoTracks->GetAt(0)->GetEncodingProperties();
		
			ENQUEUE_UNIQUE_RENDER_COMMAND_THREEPARAMETER(InitMovieTexture,
			TSharedPtr<FUWPMovieStreamer>, ThisStreamer, SharedThis(this),
			uint32, Width, VideoProperties->Width,
			uint32, Height, VideoProperties->Height,
			{
				ThisStreamer->InitMovieTexture_RenderThread(Width, Height);
			});
	}

PLEASE HELP