microsoft/AirSim #2349: Unity: Drone goes through ground and walls after continuous collisions

PaigeDavid, 2019-12-17:
Windows 10, Unity version 2018.2.14: if the drone is flying into a wall (tested on 1 meter thickness), it will eventually fly through the wall after just a few collisions. For the ground, if it comes down at any angle with a small amount of speed, it will fall through the terrain. This happens both with a controller and using the C++ API. I was unable to recreate this in Unreal as the drone does not go through the terrain or objects. 

I have looked into this a little bit and I have tried changing the Fixed Timestep, I have tried the "DontGoThroughThings" script since this sounded similar, and I have tried changing the Interpolate / Collision Detection settings on the rigidbody but that usually messes up how the drone flies.

emilykjensen, 2022-04-26:
I am having the same problem, using Windows 11 and Unity version 2019.3.12f1. The standard generated logs in Unity show that it is detecting the repeated collisions, but the drone will still fly through objects. I have also tested this on objects at least 40 meters thick, using the same material and settings as the objects provided in the demo environment. Beyond the attempted fixes listed above, increased the size of the box collider for the drone body to extend slightly beyond the rotors, but with no effect.

buvnswrn, 2022-10-22:
is this issue fixed or is any workaround available? It was working with version 2019.3.12f1 and wanted to upgrade to 2020.3.9f1 and this issue occurred to me.

buvnswrn, 2022-10-22:
> is this issue fixed or is any workaround available? It was working with version 2019.3.12f1 and wanted to upgrade to 2020.3.9f1 and this issue occurred to me.

For the Quadcopter object, a simple reset on the Rigidbody component according to https://forum.unity.com/threads/colliders-not-working-in-build-since-last-update.828468/ and undoing it or entering the values manually worked for me and the quadcopter did not fall through the floor.

buvnswrn, 2022-12-27:
I recently noticed that the issue persists when the distance between the drone and the ground is very less. If I keep more distance, then the drone lies on the ground

Kinggrass, 2023-07-03:
Any news here? Also on the same issue currently.