microsoft/AirSim #2353: How to fuse multiple imu sensors?

sauranildebs, 2019-12-20:
I am trying to simulate a drone having multiple IMU sensors on-board. It is possible to fuse multiple LIDARs on a single drone, but I am not sure if multi-IMU integration is possible. I need to collect these IMU data using ROS topics so that I can use an Extended Kalman Filter for estimation. Moreover, where can I find the IMU-specific parameter list?

marcociara379, 2019-12-24:
Hi @sauranildebs,
I guess you could try manually adding new IMUs following the structure and the content of the files located [here](https://github.com/microsoft/AirSim/tree/master/AirLib/include/sensors/imu). 
Furthermore, in [AirSimSettings.hpp](https://github.com/microsoft/AirSim/tree/master/AirLib/include/common/AirSimSettings.hpp), you might want to include your new sensor. 

On the other hand, if your IMU can be modelled as the one in airsim, but only with different parameters, I guess you can specify it directly in the `settings.json`., adding many of them to your vehicle.  Unfortunately, I do not have a computer with AirSim installed at the moment to check the details.

sauranildebs, 2020-01-07:
Hi @marcociara379 
Thank you so much for the information. I'll try adding a new IMU AirSimSettings.hpp file. I tried adding two IMUs with different name parameter on the same vehicle by specifying them in the settings.json file, but it seems I am receiving the data from only one IMU because I haven't given any parameters. Can you share with me the IMU-specific parameters such that I can edit the settings.json file accordingly for both the IMUs.

closed 2020-03-10 by ironclownfish