microsoft/AirSim #2355: AirSim C++ API Build Issue: unresolved external symbol

teddy1496, 2019-12-20:
I am trying to write a C++ API with AirSim. I followed the steps mentioned in
```
https://microsoft.github.io/AirSim/docs/apis_cpp/
```
I add the include in my c++ file 
```
#include "vehicles/car/api/CarRpcLibClient.hpp"
```

After which I added all the mentioned include and Lib Folders to my Project but I still get an LNK2019 error unresolved external symbol 

### Errors 
I will attach here the errors I receive
```
* Error LNK2019 unresolved external symbol "public: void __cdecl msr::airlib::RpcLibClientBase::confirmConnection(void)" (?confirmConnection@RpcLibClientBase@airlib@msr@@QEAAXXZ) referenced in function subscriber_main

* Error LNK2019 unresolved external symbol "public: void __cdecl msr::airlib::RpcLibClientBase::enableApiControl(bool,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?enableApiControl@RpcLibClientBase@airlib@msr@@QEAAX_NAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z) referenced in function subscriber_main

* Error LNK2019 unresolved external symbol "public: __cdecl msr::airlib::CarRpcLibClient::CarRpcLibClient(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,unsigned short,float)" (??0CarRpcLibClient@airlib@msr@@QEAA@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@GM@Z) referenced in function subscriber_main

* Error LNK2019 unresolved external symbol "public: void __cdecl msr::airlib::CarRpcLibClient::setCarControls(struct msr::airlib::CarApiBase::CarControls const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?setCarControls@CarRpcLibClient@airlib@msr@@QEAAXAEBUCarControls@CarApiBase@23@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z) referenced in function subscriber_main

* Error LNK2019 unresolved external symbol "public: virtual __cdecl msr::airlib::CarRpcLibClient::~CarRpcLibClient(void)" (??1CarRpcLibClient@airlib@msr@@UEAA@XZ) referenced in function subscriber_main	
```

I am new to developing on Windows and unclear what exactly might be the issue here since I have added the Mentioned Library and Directory Paths. Any Help would be much appreciated.

### AirSim Version Used
* I am using AirSim Version 1.2

changh95, 2019-12-26:
What's your linker configuration?

I'm not having those errors, and I'm currently using the configuration below.

**Additional Include Directories:**
`...\AirLib\deps\rpclib\include
...\AirLib\include
...\AirLib\deps\eigen3
include`

**Additional Library Directories:**
`...\AirLib\lib\x64\Debug
...\AirLib\deps\rpclib\lib\x64\Debug
...\AirLib\deps\MavLinkCom\lib\x64\Debug`

**Additional Dependencies:**
`rpc.lib`

obitodaitu, 2019-12-26:
I have same problem ,if you sovle this problem ,please tell me ,thanks.

teddy1496, 2020-01-13:
@changh95 I have a similar linker configuration, but the error still persists for me.

teddy1496, 2020-01-14:
> I have same problem ,if you sovle this problem ,please tell me ,thanks.

 @obitodaitu I was able to compile the code by adding "Airlib.lib" to "Additional Dependencies"

saihv, 2020-01-20:
Closing as the issue seems to be resolved.

closed 2020-01-20 by saihv