microsoft/AirSim #2358: Semantic segmentation colours not in the seg_rgbs.txt

ivannson, 2019-12-25:
I am collecting some segmentation images, but upon displaying them, I found that the colours in my segmentation image do not correspond to any colours specified in `seg_rgbs.txt`. 

This is the image produced:
[image]

Data collection code snippet: 
```python 
responses = client.simGetImages([
            airsim.ImageRequest("2", airsim.ImageType.DepthPerspective, compress = False, pixels_as_float = True),  #depth visualization image
            airsim.ImageRequest("0", airsim.ImageType.Scene, compress = False), #png format
            airsim.ImageRequest("0", airsim.ImageType.Segmentation, False, False)])

filename = 'images/segmentation/' + 'segmentation' + str(file_sequence)
        img1d = np.fromstring(responses[2].image_data_uint8, dtype=np.uint8) #get numpy array
        img_rgba = img1d.reshape(responses[2].height, responses[2].width, 3) #reshape array to 4 channel image array H X W X 3
        img_rgba = np.flipud(img_rgba) #original image is fliped vertically

        np.save(filename, img_rgba)
```

By looking at the colours, for example top left quadrant with a building in dark yellow, I get `[203 174  42]` as the rgb values, however this colour doesn't seem to correspond to any label `seg_rgbs.txt`.

TDYbrownrc, 2020-02-21:
I am having similar issues. What version of Unreal engine are you using? I have seen this behavior on 4.19 and 4.23.

ivannson, 2020-02-21:
@TDYbrownrc I'm using 4.22. 

The work around that I'm using right now is using infrared camera instead of segmentation view.

averyehorvath, 2020-08-14:
I pulled in the latest changes of Airsim and updated to Unreal 4.24 and that fixed it for me

jonyMarino, 2020-08-17:
@ivannson could you check?

rajat2004, 2020-08-17:
https://github.com/microsoft/AirSim/pull/2845 fixed this problem

closed 2020-08-18 by jonyMarino

lucafei, 2020-10-31:
hello, @ivannson do you have checked the workaround #2845? For me i have also met this problem that colors in my segmentation image are not same with the RGB value specified in seg_rgbs.txt.