microsoft/AirSim #2361: Pan tilt on the Drone

xlong0513, 2019-12-27:
Hi, i want to adding pan-tilt on the drone to achieve a down-looking camera to capture images. 
I have got some inspirations from [#539](https://github.com/microsoft/AirSim/issues/539). First, i have set a camera 3(bottom_center) in setting files; second i use `simSetCameraOrientation`  to achieve a the pan tilt. The attitude angles are all set as 0 to simulate down-looking effects.
Part of the codes are as following screenshot.
[image]

However, the captured images are still shaking with the drone?
Can anyone tell us what's the matter?

ACLeighner, 2020-07-17:
Hey I have a fix for the shaking drone, check it out over at:
https://github.com/microsoft/AirSim/issues/1394#issuecomment-629936379

currently running into some weird issues with set camera orientation jumping around

nethtrick, 2020-07-20:
Hello xlong0513,

I too have run into jittery camera issues - even after inverting the drone orientation. I have posted about it here:

https://github.com/microsoft/AirSim/issues/2595

I have not yet tried ACLeighner's solution but it sounds like they are still experiencing camera jitter issues. Hopefully between us we can figure this out!

xlong0513, 2020-08-04:
@ACLeighner @nethtrick  Hi, there, thanks for the reply. However, I am now isolated at home now, and cannot work on it. When back to work, I will try the solution and reply.