microsoft/AirSim #2364: PositionController does not support  yaw axis i.e. 3

jinkman, 2019-12-30:
[Spawn vehicle via rpc](https://github.com/microsoft/AirSim/pull/1937/commits/dc8fc2cbae4c9224604472f1ede353587eec9100),
I made a follow this link to generate the vehicle's command, this command did a good job, 
but I use the client. The moveToPositionAsync (10, 10, 10, 10, 10, DrivetrainType: : ForwardOnly, YawMode (false, 0.0), 1.0, 1.0, "My") - > waitOnLastTask ();",
there will be mistakes. Like This:
[image]
The vehicle generated from the.json file is fine.I'd like to know which circumstances led to the error.The error seems to be related to the length of time.My game mode is sampleflight

jinkman, 2019-12-30:
LogTemp: Error: Exception occurred while updating world: PositionController does not support yaw axis i.e. 3
LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()

jonyMarino, 2020-09-09:
treated on new PR #2390. Look there

closed 2020-09-09 by jonyMarino