microsoft/AirSim #2367: The Release doesn't work, and the LOG shows "LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)"

DinghaoZhang, 2020-01-07:
AirSim is great, but I can't run the Release - "Zhangjiajie" and "SubT". the LOG file shows as following.
``` bash
LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
LogWindows: File 'aqProf.dll' does not exist
LogWindows: Failed to load 'VtuneApi.dll' (GetLastError=126)
LogWindows: File 'VtuneApi.dll' does not exist
LogWindows: Failed to load 'VtuneApi32e.dll' (GetLastError=126)
LogWindows: File 'VtuneApi32e.dll' does not exist
```
My operation system is Win10, and I use Unreal 4.18.3, VS2017 with "Windows 8.1 SDK".
I try to run it with administrator, but doesn't work.

lucaspojo, 2020-03-05:
Up ? Same !

WLB1996, 2020-05-20:
same problem ,did you solve it?

jonyMarino, 2020-09-04:
Closed due to age.
Please feel free to open a new issue (refering this one) if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-09-04 by jonyMarino

bkbbq, 2022-05-05:
so what the really problem is and how to fix??? have stuff in this case for a few days when running with ue5 git source code . thanks for reply...

jonyMarino, 2022-05-05:
@bkbbq you can open a new issue, but we don't officially support ue.