microsoft/AirSim #2369: Running SLAM system on AirSim

xuhao1, 2020-01-08:
Hi Guys,
I have successfully run a well-known visual-inertial system VINS(https://github.com/HKUST-Aerial-Robotics/VINS-Fusion) on AirSim (which is amazing good!) based on the ros package which verified the possibility of simulating my own aerial swarm in airsim.

However, I found few issues about the ros package;

1. The Unit of the angular velocity is wrong. The data from airsim is already rads
2. The coordinate system seems is confused
3. The process lock lower the whole  performance, with the lock now we can only get IMU for ~20hz
4. The ros package used simGetImages with single request a few times, which will be slow when we use a stereo Camera.
5. The ros package use __soft time stamp__ (the receive ros::Time::now) instead simulation time
6. Nowhere to set host ip
For problem 1. 5. 6. I have submit a pull request #2368,
For problem 234, my modification loses the generic but it solves the problem, gives me 20HZ stereo images and 300hz IMU on a remote notebook and successfully running VINS-Fusion in AirSim, here is my repo https://github.com/xuhao1/airsim_djisdk. If you interesting running VINS or other slam methods, you can have a try

simengangstad, 2020-01-09:
We've worked some with VINS in AirSim as well, but haven't gotten any good results. Do you have any numbers on how much the position drifts compared to ground truth from your current setup?

xuhao1, 2020-01-10:
@simengangstad , there is drift. However, drift is normal for VINS-Fusion without Loop Closure on. If you are interested in it, you can have a try with my package.

Beniko95J, 2020-02-20:
Hi, may I ask how did you achieve 20Hz stereo images? I am also trying to run [VINS-Mono](https://github.com/HKUST-Aerial-Robotics/VINS-Mono) on AirSim, and when I set the image resolution to 288x512 (original 144x256) and set the timer to save images every 0.05 second, I can only obtain 3Hz 288x512 images through its [image apis](https://microsoft.github.io/AirSim/docs/image_apis/). Do you have any idea about this?

kminoda, 2020-05-01:
I am having similar issue in similar situation, which is to run VIO algorithm on AirSim ros pkg. I want to change frame rate of camera and IMU. 

I assume that its "update_airsim_img_response_every_n_sec" in [here](https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_ros_pkgs/launch/airsim_node.launch) that corresponds to camera frame rate. But it doesn't make any difference. Camera frame rate is still 3Hz even though I set the value to 0.1 (so it should be 10Hz).

I modified airsim_ros_wrapper.cpp according to [this repos](https://github.com/xuhao1/airsim_ros_pkgs/tree/c1bb1579b64299cd987f25e11c01f061170e60a3), but it doesn't solve the problem.

Does anyone know how to solve this problem?

dssdyx, 2020-05-11:
@xuhao1 Which coordinate did you transform imu link to? NED to ENU or other? I see your body_T_cam in vins yaml, is corresponding to ENU. The linear vel from NED to ENU should be (x,y,z)->(x,-y,-z),but your code is (x,y,z)->(-x,y,-z). I feel really confused.

manthan99, 2020-05-12:
I am also having a similar issue with the published rate. I tried modifying the code according to [this repo](https://github.com/xuhao1/airsim_ros_pkgs/tree/c1bb1579b64299cd987f25e11c01f061170e60a3) but I am only getting a publish rate of around 5 Hz for the camera and 10 Hz from IMU. Changing the "update_airsim_img_respone_every_n_sec" doesn't seem to increase the rate. 
Any solution to increase the rate?

I am having i7-6700HQ, 16 Gb ram and Gtx 1070

rajat2004, 2020-05-12:
Could you check the parameter `DrawDebugPoints` in Lidar in the settings.json? If `true`, that causes a huge slowdown in the current latest binaries. Fix #2614 has been merged into master, but the problem is there in the binaries. Just change that to `false` and should improve the rates
There's also another WIP PR #2472 to increase the camera rate

manthan99, 2020-05-12:
I am not using a Lidar. Here is the settings file which I am using [[Settings.json]](https://github.com/xuhao1/airsim_ros_pkgs/blob/master/vins_airsim/settings.json)

Though the rate significantly improved to 20 Hz upon using the "NoDisplay" viewmode

rajat2004, 2020-05-12:
Mentioned about the Lidar since that was causing problems with people trying out the ROS package tutorial and all. Using NoDisplay will definitely improve the rate since rendering is less
@xuhao1 It would be great to have a PR with your changes if you're interested

manthan99, 2020-05-12:
There seems to be a problem with the sync of images from left and right cameras ( ~0.05 sec) 
@xuhao1 did you face this issue?

kminoda, 2020-05-13:
@manthan99 It’s nice to hear that you managed to get images on 20 Hz by using NoDisplay mode.
Did you just change “ViewMode” to NoDisplay in the settings.json you posted?

manthan99, 2020-05-13:
@kminoda Yes I justed changed the "ViewMode" to NoDisplay in the settings.json file. In addition, I also replaced the [airsim_node.cpp ](https://github.com/xuhao1/airsim_ros_pkgs/blob/c1bb1579b64299cd987f25e11c01f061170e60a3/src/airsim_node.cpp)file that is given in this repo though I do not know if that made a difference.

kminoda, 2020-05-20:
With an advice from a researcher who previously worked on running VIO in AirSim, I managed to run VINS-Mono on AirSim by changing "ClockSpeed" in settings.json. 
For me, since I could get 3Hz at maximum, I made ClockSpeed to 0.033 so that I can get 100Hz (in simulation time, not in actual time) for IMU.
OFC it take much longer time to collect data even for short ones, but I can now estimate the pose of UAV without any drift.

manthan99, 2020-05-20:
@kminoda Di you change the coordinate system for IMU ? And which parameters file did you use ?

kminoda, 2020-05-20:
@manthan99  I didn’t change IMU frame (but maybe it’s better to do so).
I verified that VINS-Mono is working by comparing the odometry output from vins and odom_local_ned. I changed the frame of odom_local_ned in comparing them.

davdmc, 2021-06-22:
Since airsim_ros_wrapper changed quite a lot since this post, the referenced repos are old for being used as reference. I have added a timer to publish the IMU at a larger frequency in the main thread (as opposed to images and Lidar) [link](https://github.com/davdmc/AirSim). I don't know if this is a good solution but still I'm not able to use VINS-Fusion. I think that now the problem is related to synchronization or the movement I'm using. Main complaints from VINS:

[ INFO] [1624356525.747085129]: misalign visual structure with IMU
[ INFO] [1624356525.863072756]: Not enough features or parallax; Move device around

I am still having the problems of stereo pair synchronization and it is not clear for me if the coordinate system confusion is fixed in later versions of AirSim. Maybe the users who previously answered could provide updated answers or hints about the specific changes in the ROS wrapper code. Also, is there any intention of considering this issue in future versions of the official repo? I am willing to contribute :)

phil333, 2021-06-23:
@davdmc 
That sounds like you having trouble configuring Vins-fusion? i needs a correct transformation matrix between the imu and each camera. If thats not properly set. Secondly, what environment are you using? is empty or are there any things that can be tracked
I stopped going down this road because the left and right image are not synchronized. As long as this is not sorted, testing VO or VIO objectively is difficult, especially when producing larger images like for example the ZED camera. Essentially the images should be captured in parallel, but this might require freezing the sim, capturing and then unfreezing (not an expert on this).

xxEoD2242, 2021-06-23:
@davdmc 

I'll give you some general tips on how to utilize this:

1. You need to rewrite the image request structure of the ROS wrapper. Currently, the ROS wrapper is not written efficiently. I would recommend building your own thread that only gets the stereo images as a pair in the image request.
2. You need to increase the tolerance of VINS Fusion. The value is currently 0.03 but you need to increase that to 0.07.
3. Most likely your timestamps are not working properly for the images. I would look at those stamps and make sure they are in ROS time.

Just general pointers for working with VIO systems in AirSim. Happy to help with more specific issues.

phil333, 2021-06-23:
@xxEoD2242 
when you say tolerance, which value in vins fusion are you talking about?
https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/master/config/euroc/euroc_stereo_imu_config.yaml

GimpelZhang, 2021-06-24:
> @xxEoD2242
> when you say tolerance, which value in vins fusion are you talking about?
> https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/master/config/euroc/euroc_stereo_imu_config.yaml

I think @xxEoD2242 means the sync tolerance in the code processing stereo images: [rosNodeTest.cpp](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/master/vins_estimator/src/rosNodeTest.cpp#L85-L95)

phil333, 2021-06-24:
ok. There is a good reason for this tolerance to be so low, and that it is hard coded. But i guess you can test if there are any other issues. In that case i would use the car model for testing, as it can move slower.

xxEoD2242, 2021-06-24:
@phil333 @GimpelZhang  Yup, that was what I was talking about.

From my experience with AirSim, even with every optimization, you aren't going to get that time difference to be within Fusion tolerance. If AirSim is updated with faster image capture internally, then that is possible. But from how AirSim works under the hood, getting UE to actually generate the images is just not fast enough for 3 milliseconds of latency.

phil333, 2021-06-24:
:(

NickPerezCarletonUniversity, 2021-07-08:
I understand there is something fundamentally wrong with how the images are captured. But I don't see how setting the clock speed very low in simulation doesn't give an acceptable delay between when left images are taken and right images are taken. doesn't AirSim also have a RGBD camera sensor? That is ultimately quite similar to what stereo cameras have to offer.

xxEoD2242, 2021-07-08:
@NickPerezCarletonUniversity if that works, then great. However, when working with Fusion, we raise the tolerance of the image difference. For our purposes, lowering the clock speed isn't an option due to processing requirements and time calculations.

NickPerezCarletonUniversity, 2021-07-08:
"For our purposes, lowering the clock speed isn't an option due to processing requirements and time calculations." what do you mean by this? All I am saying is you can reduce how fast the simulation environment runs. https://microsoft.github.io/AirSim/settings.html#clockspeed

xxEoD2242, 2021-07-08:
Reducing the clock speed means that the environment takes longer to update. We do real time trajectory generation that requires feedback that is close to what we do in the real world. Great for developing VIO but not for implementing on real world systems.

NickPerezCarletonUniversity, 2021-07-09:
@phil333 you can freeze and unfreeze the simulation using simPause https://microsoft.github.io/AirSim/image_apis.html#getting-images-with-more-flexibility

I emailed the creators of the TartanAir dataset https://theairlab.org/tartanair-dataset/ and they said this is what they did to get around the left and right camera synchronization issue.

phil333, 2021-07-12:
@NickPerezCarletonUniversity thx for the input. I also assume this is what needs to be done with the current airsim-rosbridge to get this working. I currently dont have the time to look into this. For a SLAM system, the delay is not necessarily a problem. It is also possible to record the data and build a rosbag with the correct timestamp instead. As we are talking about VIO data, i would also still want the IMU data generated. Is this something that can be retrieved with the API?

NickPerezCarletonUniversity, 2021-07-12:
@phil333 yes, you can definitely record IMU data! I'm not sure what the exact API call is.

NickPerezCarletonUniversity, 2021-07-13:
@phil333 yes an offline rosbag-bridge would be useful. if you get around to it before I do, do you mind sharing? I think Gazebo is a good simulator as well (https://www.youtube.com/watch?v=UtJFUeHXH6I&lc=UgzVgecehqw-QEwCXj94AaABAg.9Pa2S1Iossx9Pa2nEbmfCG). Not sure if it offers IMU support. you might still be able to simulate real-time SLAM while pausing and unpausing AirSim quickly.

phil333, 2021-07-14:
@NickPerezCarletonUniversity i might look into this after i finish my thesis, but not a priority right now. Gazebo is a good simulator and well integrated into ROS, but for computer vision tasks, it is just not comparable to real data. If that works for you, then you should avoid Airsim and stick to gazebo, no question. Airsim was designed for drones, which generally don't have a stereo camera setup. For such a configuration, the simulator is perfectly fine, but i agree, this issue is a major issue.

xxEoD2242, 2021-07-14:
@phil333 @NickPerezCarletonUniversity

For stereo vision, if you are proficient in threading in C++, it is entirely possible to use ROS to generate stereo images as a pair. I would avoid the provided AirSim ROS wrapper, however. That system is not built with performance in mind, generally because it processes each camera frame as a separate image request. You can, however, use the C++ API provided to build your own ROS wrapper that has minimal features.

From that standpoint, you can then write 2 threads, one for IMU and one for the stereo images. From our work, I would highly recommend looking at synchronization policies to ensure that you have a fixed time difference between the IMU data and the images. Because of variable performance from AirSim this will destroy performance of a VIO algorithm, as the time differences shift constantly. You can use adaptive algorithms (which are in the sync policies in ROS) to mitigate some of this but everything has tradeoffs.

Sorry that this is really a "You have to do a bunch of work yourself" answer but that's what we have done to get our systems to work.

NickPerezCarletonUniversity, 2021-07-14:
@xxEoD2242 can you share the code to your solution? would be really helpful.

hipforth, 2021-09-07:
> @xuhao1 Which coordinate did you transform imu link to? NED to ENU or other? I see your body_T_cam in vins yaml, is corresponding to ENU. The linear vel from NED to ENU should be (x,y,z)->(x,-y,-z),but your code is (x,y,z)->(-x,y,-z). I feel really confused.

NED to ENU :  I think it should be (x, y, z) -> (y, x, -z),  why  (x,y,z)->(x,-y,-z)?

NickPerezCarletonUniversity, 2021-09-13:
@xxEoD2242 any progress on sharing your ROS wrapper code?

xxEoD2242, 2021-09-21:
@NickPerezCarletonUniversity 

I have been swamped with other projects. If you want to a swipe at it, check out my ROS wrapper:

[AirSim ROS Wrapper](https://github.com/Green-Team-Systems/AirSim/tree/vins_mono_setup/ros/src/airsim_ros_pkgs)

This assumes that you have the following in your settings.json file:

```
Drone Name is PX4

You have 1 camera called `front_center_custom` that is a DepthPerspective Camera (option 2).
You have 1 camera called 'front_center' that is a scene camera.
```

The way that this works is that it builds a thread for each image. So, you wanted stereo images, you will need to modify the scene image to be two images. Then, on line [1496](https://github.com/Green-Team-Systems/AirSim/blob/663556eadbd39c35eba59c5fe201f80092fd2526/ros/src/airsim_ros_pkgs/src/airsim_ros_wrapper.cpp#L1596), you need to modify the AirSim request to be two scene images.

I'll get around to refactoring this ROS wrapper to remove all of the other stuff and slim down state updates as well. Ask any questions that you may have about how to set this up.

lnexenl, 2022-02-25:
> I also made it work by collecting IMU data with ROS while using computer vision mode for camera images (This way you can get completely synchronized images).
> 
> Here is a procedure.
> 
> * Use airsim_ros_pkg to record IMU and ground truth odometry msgs to generate a rosbag.
> * Set AirSim to computer vision mode.
> * Capture images from the point which you can get from gt odometry in the rosbag, and write these into it as rosmsg.
> 
> Hope it will help someone :)

@kminoda Thanks! This definitely works well! I recorded a stereo camera rosbag, and run VINS-Fusion with it. Finally I get a perfect result. 

And I recommend everyone who wants to get synchorized image pairs to use this method:

1. Use airsim_ros_pkgs to record vehicle groundtruth pose and IMU data in any mode you want.
2. Switch to **ComputerVision** mode, add cameras you want to record.
3. Read rosbag, get recorded vehicle pose from it. Use AirSim API to set vehicle pose, and use AirSim **C++ API** to get images and write images back to rosbag. (I found that Python API is much more slower than C++ API)

hoangvietdo, 2022-06-14:
> 

Could you provide more detailed steps about this procedure?

hoangvietdo, 2022-06-14:
> @kminoda Thanks! This definitely works well! I recorded a stereo camera rosbag, and run VINS-Fusion with it. Finally I get a perfect result.
>
> And I recommend everyone who wants to get synchorized [...]

Could you provide more detailed steps about this procedure?

lnexenl, 2022-08-04:
> Could you provide more detailed steps about this procedure?

sorry for a late reply, I have written a blog post and maybe you can refer to that: [Record Synchronized Image Pair with AirSim](https://blog.lnex.dev/en/posts/airsim_rec/)