microsoft/AirSim #2370: libUE4Editor-Blocks.so and libUE4Editor-AirSim.so are missing or built with a different engine version

Neabfi, 2020-01-08:
Hello,

I try to run the Block environment on a headless server. I followed all the steps, I’m now trying to run the UE4Editor binary.
`./UE4Editor /home/neabfi/AirSim/Unreal/Environments/Blocks/Blocks.uproject`

I have an error that says that "libUE4Editor-AirSim.so” and “libUE4Editor-AirSim.so” are missing or built with a different engine version. I tried to rebuild them but I stil get the same error. Since I’m on a headless server I can’t do what is written in the doc: “If you get prompted for incompatible version and conversion, select In-place conversion which is usually under "More" options. If you get prompted for missing modules, make sure to select No so you don't exit."

Thank you!

```
Increasing per-process limit of core file size to infinity.
LogInit: Display: Running engine for game: Blocks
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(-502864963) SRandInit(-502864963).
LogInit: Read version file with build ID 'd38ec5f6-5194-451d-85c6-f3c2f7c47d91'
LogInit: Registered custom module enumerator with build ID 'd38ec5f6-5194-451d-85c6-f3c2f7c47d91'
LogTaskGraph: Started task graph with 5 named threads and 38 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.061077
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin AndroidDeviceProfileSelector
LogPluginManager: Mounting plugin GoogleCloudMessaging
LogPluginManager: Mounting plugin WindowsMoviePlayer
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin AppleARKit
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin IOSDeviceProfileSelector
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Plugin shader directory ../../../Engine/Plugins/Runtime/Oculus/OculusVR/Shaders found
LogPluginManager: Mapping shader source directory ../../../Engine/Plugins/Runtime/Oculus/OculusVR/Shaders to virtual directory /Plugin/OculusVR
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin AndroidMoviePlayer
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin AppleMoviePlayer
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin AlembicImporter
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin HTML5Networking
LogPluginManager: Mounting plugin LevelSequenceEditor
LogPluginManager: Mounting plugin ActorSequence
LogPluginManager: Mounting plugin MatineeToLevelSequence
LogPluginManager: Mounting plugin SpeedTreeImporter
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin MobileLauncherProfileWizard
LogPluginManager: Mounting plugin AudioCapture
LogPluginManager: Mounting plugin GameplayTagsEditor
LogPluginManager: Mounting plugin PluginBrowser
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemIOS
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemGooglePlay
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin VisualStudioCodeSourceCodeAccess
LogPluginManager: Mounting plugin VisualStudioSourceCodeAccess
LogPluginManager: Mounting plugin PerforceSourceControl
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin XCodeSourceCodeAccess
LogPluginManager: Mounting plugin KDevelopSourceCodeAccess
LogPluginManager: Mounting plugin NullSourceCodeAccess
LogPluginManager: Mounting plugin SubversionSourceControl
[... 101 lines cut ...]
Would you like to rebuild them now?: Missing Blocks Modules: No is implied.
LogExit: Preparing to exit.
LogModuleManager: Shutting down and abandoning module AnimationModifiers (34)
LogModuleManager: Shutting down and abandoning module PropertyEditor (33)
LogModuleManager: Shutting down and abandoning module AudioEditor (30)
LogModuleManager: Shutting down and abandoning module TextureCompressor (28)
LogModuleManager: Shutting down and abandoning module ShaderCore (26)
LogModuleManager: Shutting down and abandoning module Landscape (24)
LogModuleManager: Shutting down and abandoning module SlateRHIRenderer (22)
LogModuleManager: Shutting down and abandoning module OpenGLDrv (20)
LogModuleManager: Shutting down and abandoning module AnimGraphRuntime (18)
LogModuleManager: Shutting down and abandoning module Renderer (16)
LogModuleManager: Shutting down and abandoning module Engine (14)
LogModuleManager: Shutting down and abandoning module CoreUObject (12)
LogModuleManager: Shutting down and abandoning module NetworkFile (10)
LogModuleManager: Shutting down and abandoning module CookedIterativeFile (8)
LogModuleManager: Shutting down and abandoning module StreamingFile (6)
LogModuleManager: Shutting down and abandoning module SandboxFile (4)
LogModuleManager: Shutting down and abandoning module PakFile (2)
LogExit: Exiting.
LogInit: Tearing down SDL.
Exiting abnormally (error code: 1)
```

I’m on Ubuntu 18.04.2 LTS

rajat2004, 2020-01-16:
Maybe you could try something like [X Windows Forwarding](https://unix.stackexchange.com/questions/12755/how-to-forward-x-over-ssh-to-run-graphics-applications-remotely) if you're ssh'ing into the system? That should probably show the prompt
There's support for running AirSim inside docker, see https://github.com/Microsoft/AirSim/blob/master/docs/docker_ubuntu.md
Info about this is there in other issues as well

Neabfi, 2020-01-16:
I partially fixed the issue by using the v1.2.0 release for Linux instead. The only drawback is that the release is old, and so I can’t use the new apis.

madratman, 2020-04-14:
new linux releases were made.

closed 2020-04-14 by madratman

tanyaapaul, 2022-10-05:
How did the issue got fixed? I am having the same problem / error message