microsoft/AirSim #2371: Is it possible to steer car in AIRSIM by providing input from CSV file which has steering angle, velocity, brake, gear and timestamp.

surendrasah, 2020-01-10:
Hi Team,
I am using AIRSIM simulator in linux and I have a CSV file which has necessary parameter (velocity, steering angle, brake, gear and timestamps) required to steer the vehicle. Any help on this would be really appreciated.

saihv, 2020-01-20:
You can iterate through the CSV (using Python for example) and send steering angle, throttle/brake commands for each line through the setCarControls() function.

closed 2020-01-21 by saihv

surendrasah, 2020-01-24:
Hi Saihv,
Thanks for the response. Yeah, iteration can work with modifications to the values.
But i still have few doubts.
I have four brake value from four tires. how do i convert it to a single value as AIRSIM expects one.
I have steering angle but its in positive sign like 20, 30, 120 etc. degree. how would i calculate if its turn left or right. Any idea on this would be really appreciated.