microsoft/AirSim #2373: PythonClient/Multirotor Scripts do not work together with PX4

gonzaloaguilarjimenez, 2020-01-12:
Hello @lovettchris 

Sorry to bother you, but I hope you can give me a solution.

firstly, I will introduce my versions:

- OS: Ubuntu 16.04
- AirSim: 1.2
- Python version: 2.7
- Unreal Engine: 4.18

The seetings.json I am using is as follows: 

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2, 
  "CameraDefaults":{
	"CaptureSettings":[
		{
		"ImageType": 4,
		"Width": 256,
		"Height": 144,
		"FOV_Degrees": 90,
		"AutoExposureSpeed": 100,
		"MotionBlurAmount": 0
		}
	]
  },
  "SimMode": "",
  "LogViewerHostIp": "",
  "Vehicles":{
	"PX4":  {
		"VehicleType": "PX4Multirotor",
		"LogViewerHostIp": "",
		"UseSerial": false
		}
	},
   "SubWindows": [
    {"WindowID": 0, "CameraName": "0", "ImageType": 4, "Visible": false},
    {"WindowID": 1, "CameraName": "0", "ImageType": 7, "Visible": false},
    {"WindowID": 2, "CameraName": "0", "ImageType": 0, "Visible": false}    
  ],

   "Recording": {
    "RecordOnMove": true,
    "RecordInterval": 0.05,
    "Cameras": [
        { "CameraName": "0", "ImageType": 4, "PixelsAsFloat": false, "Compress": true }
    ]
  }

}

With this configuration PX4 works perfectly but there is no way to launch a for example hello_drone.py, however if the configuration for the settings.json is as follows then then obviously PX4 is deactivated, but the PythonClient/Multirotor scripts work without any type of problem.

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2, 
  "CameraDefaults":{
	"CaptureSettings":[
		{
		"ImageType": 4,
		"Width": 256,
		"Height": 144,
		"FOV_Degrees": 90,
		"AutoExposureSpeed": 100,
		"MotionBlurAmount": 0
		}
	]
  },
  "SimMode": "",
  "LogViewerHostIp": "",

   "SubWindows": [
    {"WindowID": 0, "CameraName": "0", "ImageType": 4, "Visible": false},
    {"WindowID": 1, "CameraName": "0", "ImageType": 7, "Visible": false},
    {"WindowID": 2, "CameraName": "0", "ImageType": 0, "Visible": false}    
  ],

   "Recording": {
    "RecordOnMove": true,
    "RecordInterval": 0.05,
    "Cameras": [
        { "CameraName": "0", "ImageType": 4, "PixelsAsFloat": false, "Compress": true }
    ]
  }

}

My aim is to use both PX4 and Scripts together and everything I found about this is the issue #1361 but it does not mention what changed or how to solve it.

I hope you can give me any idea to go on.

Thanks a lot in advance.

Best regards.

lovettchris, 2020-01-12:
Depending on the version of PX4 you are using you might be running into the lockstep issue, see recently resolved [PR # 2310](https://github.com/microsoft/AirSim/pull/2310).

gonzaloaguilarjimenez, 2020-01-15:
Hello, 

It seems it can be that since I have the PX4 version 1.0. 

However, I am not able to install the current version using the documentation in "Setting up PX4 Software-in-Loop" from Airsim. 

I will try it deeply and inform you if that was finally the error so that the issue can be closed.

Thanks again.

ibrahim0v, 2020-01-16:
I can assure that it works on HITL with quite the simplest setups as mentioned in the documents

ibrahim0v, 2020-01-16:
It works both in HITL and SITL modes + HITL tested with PX2Cube and Pixhawk1 mRo

gonzaloaguilarjimenez, 2020-01-18:
Hello @ibrahim0v 

Which version of PX4 do you have? I have currently updated to V1.8.2 and it does not work for me.

Still, once I launch PX4, none of the multirotor scripts can connect. 

Thanks in advance.

Best regards.

gonzaloaguilarjimenez, 2020-01-19:
Hello again, 

I have notices that only after and Operation Timeout, the scripts can work and I get full control over QGC, PX4.

Do you have an idea why this can happen?

Thanks.

gonzaloaguilarjimenez, 2020-01-29:
Hello, 

Any response to this?

Thanks in advance.

Best regards.

lovettchris, 2020-01-29:
Can you check if the PX4 serial connection was fixed by this recent pull request?
https://github.com/microsoft/AirSim/pull/2397
And the new command line for launching SITL as documented [here](https://github.com/microsoft/AirSim/blob/master/docs/px4_sitl.md) is: 
```
make px4_sitl_default none_iris
```
Also please check that it works on AirSim without QGC.  QGC might be interfering with the connection to PX4.

gonzaloaguilarjimenez, 2020-02-01:
Hello, 

I tried what you've said and here below is what I get:

_terminating with uncaught exception of type std::runtime_error: UdpClientPort socket send failed with error: 111_

_Signal 6 caught._
_Malloc Size=131076 LargeMemoryPoolOffset=131092_
_CommonLinuxCrashHandler: Signal=6_
_Malloc Size=65535LargeMemoryPoolOffset=196655_
_Malloc Size=43133 LargeMemoryPoolOffset=239805_
_Engine crash handling finished; re-raising signal 6 for the default handler. Good bye._
_Aborted (core dumped)_

However, if I re-run  AirSim without re-running PX4 (still with the session of PX4 that threw the error) I do not receive the error 111. In this case, still I cannot make the scripts work.

I tried all this for sure using the command line for launching "make px4_sitl_default none_iris" with and without QGC, and still didn't get the expected result.

jonyMarino, 2021-06-03:
Hi @gonzaloaguilarjimenez! Do you still have this issue? Try with [our latest version](https://github.com/microsoft/AirSim/releases)

closed 2021-06-03 by jonyMarino

clovett, 2021-06-04:
Note: I have confirmed you can run 2 python scripts simultaneously using the latest bits.  This one is running `path.py` and `opencv_show.py` at the same time.  But you will have to remove this code from opencv_show.py:
```python
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)
client.takeoffAsync().join()
```
because this will interfere with the flaying that `path.py` is doing.  So long as the scripts are not confusing each other like that things would work fine.

[image]