microsoft/AirSim #2375: Error building Airsim on mac book pro Catalina Version 10.15.2

retif, 2020-01-12:
`22:04 $ ./build.sh
+ MIN_GCC_VERSION=6.0.0
+ gccBuild=false
+ [[ 0 -gt 0 ]]
+ '[' '!' -d ./external/rpclib/rpclib-2.2.1 ']'
+ '[' -d ./cmake_build ']'
++ which cmake
+ CMAKE=/usr/local/bin/cmake
+ false
+ [[ ! -d llvm-source-50 ]]
+ [[ -d ./llvm-build/output/lib ]]
+ build_dir=build_debug
++ uname
+ '[' Darwin == Darwin ']'
+ export CC=/Users/user/Documents/la/AirSim/llvm-build/output/bin/clang-5.0
+ CC=/Users/user/Documents/la/AirSim/llvm-build/output/bin/clang-5.0
+ export CXX=/Users/user/Documents/la/AirSim/llvm-build/output/bin/clang++
+ CXX=/Users/user/Documents/la/AirSim/llvm-build/output/bin/clang++
+ [[ -d ./AirLib/deps/eigen3/Eigen ]]
+ echo 'putting build in build_debug folder, to clean, just delete the directory...'
putting build in build_debug folder, to clean, just delete the directory...
+ [[ -f ./cmake/CMakeCache.txt ]]
+ [[ -d ./cmake/CMakeFiles ]]
+ [[ ! -d build_debug ]]
+ mkdir -p build_debug
+ pushd build_debug
+ /usr/local/bin/cmake ../cmake -DCMAKE_BUILD_TYPE=Debug
-- The C compiler identification is Clang 5.0.0
-- The CXX compiler identification is Clang 5.0.0
-- Check for working C compiler: /Users/user/Documents/la/AirSim/llvm-build/output/bin/clang-5.0
-- Check for working C compiler: /Users/user/Documents/la/AirSim/llvm-build/output/bin/clang-5.0 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Users/user/Documents/la/AirSim/llvm-build/output/bin/clang++
-- Check for working CXX compiler: /Users/user/Documents/la/AirSim/llvm-build/output/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Running CommonSetup...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- found AIRSIM_ROOT=/Users/user/Documents/la/AirSim
-- Performing Test UNUSED_LAMBDA_CAPTURE_WARN_SUPPORTED
-- Performing Test UNUSED_LAMBDA_CAPTURE_WARN_SUPPORTED - Failed
-- Performing Test ZERO_AS_NULL_POINTER_CONSTANT_WARN_SUPPORTED
-- Performing Test ZERO_AS_NULL_POINTER_CONSTANT_WARN_SUPPORTED - Failed
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/user/Documents/la/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/user/Documents/la/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/user/Documents/la/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/user/Documents/la/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/user/Documents/la/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/user/Documents/la/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/user/Documents/la/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/Users/user/Documents/la/AirSim
-- Configuring done
-- Generating done
[... 234 lines cut ...]
            ^
/Users/user/Documents/la/AirSim/external/rpclib/rpclib-2.2.1/dependencies/include/format.h:1292:27: note: in instantiation of member function
      'clmdep_fmt::internal::PrintfArgFormatter<wchar_t>::visit_cstring' requested here
      return FMT_DISPATCH(visit_cstring(arg.string.value));
                          ^
/Users/user/Documents/la/AirSim/external/rpclib/rpclib-2.2.1/dependencies/src/format.cc:1108:54: note: in instantiation of member function
      'clmdep_fmt::internal::ArgVisitor<clmdep_fmt::internal::PrintfArgFormatter<wchar_t>, void>::visit' requested here
    internal::PrintfArgFormatter<Char>(writer, spec).visit(arg);
                                                     ^
/Users/user/Documents/la/AirSim/external/rpclib/rpclib-2.2.1/lib/rpc/client.cc:69:20: warning: lambda capture 'max_read_bytes' is not required to be captured
      for this use [-Wunused-lambda-capture]
            [this, max_read_bytes](std::error_code ec, std::size_t length) {
                   ^
/Users/user/Documents/la/AirSim/external/rpclib/rpclib-2.2.1/lib/rpc/detail/server_session.cc:50:40: warning: lambda capture 'max_read_bytes' is not required
      to be captured for this use [-Wunused-lambda-capture]
        read_strand_.wrap([this, self, max_read_bytes](std::error_code ec,
                                       ^
18 warnings generated.
make[1]: *** [MavLinkCom/CMakeFiles/MavLinkCom.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
1 warning generated.
1 warning generated.
[ 36%] Linking CXX static library ../../output/lib/librpc.a
[ 36%] Built target rpc
make: *** [all] Error 2`

lovettchris, 2020-01-13:
Can you try [pull request 2376](https://github.com/microsoft/AirSim/pull/2376) to see if that fixes this build error?

retif, 2020-01-16:
Yes, your pull request helps. AirSim plugin is built!

retif, 2020-01-17:
Also i was trying to build AirSim Unity Wrapper and it failed to find boost headers and libraries.
AirSim/Unity/AirLibWrapper/CMakeLists.txt does not manage boost library.
Error of finding headers i managed to fix by adding 
`FIND_PACKAGE(Boost REQUIRED)
MESSAGE("Boost information:") 
MESSAGE("  Boost_INCLUDE_DIRS: ${Boost_INCLUDE_DIRS}") 
MESSAGE("  Boost_LIBRARIES: ${Boost_LIBRARIES}") 
MESSAGE("  Boost_LIBRARY_DIRS: ${Boost_LIBRARY_DIRS}") 
MESSAGE("  Boost_FILESYSTEM_LIBRARY: ${boost_filesystem_DIR}") 

INCLUDE_DIRECTORIES(${Boost_INCLUDE_DIRS})
LINK_DIRECTORIES(${Boost_LIBRARIES})`
But it still can not link with boost
`0:02 $ ./build.sh
AirSimRootdirectory:/Users/user/Documents/la/AirSim
Boost information:
  Boost_INCLUDE_DIRS: /usr/local/include
  Boost_LIBRARIES:
  Boost_LIBRARY_DIRS:
  Boost_FILESYSTEM_LIBRARY:
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/user/Documents/la/AirSim/Unity/linux-build
[  1%] Linking CXX CFBundle shared module AirsimWrapper.bundle/Contents/MacOS/AirsimWrapper
ld: library not found for -lboost_filesystem
clang-5.0: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [AirsimWrapper.bundle/Contents/MacOS/AirsimWrapper] Error 1
make[1]: *** [CMakeFiles/AirsimWrapper.dir/all] Error 2
make: *** [all] Error 2
cp: AirsimWrapper.bundle is a directory (not copied).`

I was installed boost with
`brew install boost`
and can find its library at /usr/local/lib
`10:06 $ ls -1 | grep files
libboost_filesystem-mt.a
libboost_filesystem-mt.dylib
libboost_filesystem.a
libboost_filesystem.dylib`

rajat2004, 2020-01-18:
@retif You were pretty close :) I managed to setup builds for Ubuntu 16.04, 18.04, Windows & OSX on Travis (https://github.com/microsoft/AirSim/pull/2356) Took some time and a few hacks :sweat_smile: but was able to get builds for Unreal and Unity working on all of them.
Though Travis needs to enabled for AirSim again since it's not triggering right now

I have 2 other PRs for fixing these build issues, first of which is https://github.com/microsoft/AirSim/pull/2333, which fixes some slightly subtle problems in `setup.sh` & `build.sh`
The other is https://github.com/microsoft/AirSim/pull/2328 which fixes problems for Unity build, this one has the CMake additions as well
Sorry for the spreading of fixes across different PRs, things are separate from each other so as to reduce the scope of individual PRs and for easier review and understanding. Merging them will cause conflicts in some places but that can be easily fixed one after the other.

Getting builds to pass will need all 3 PRs, but this will reduce the likelihood of things breaking in the future.
Finally, Travis build of the end result after merging everything - https://travis-ci.com/rajat2004/AirSim/builds/145052865

Hope this helps!

o0Ignition0o, 2020-01-18:
Awesome job ! I'm struggling with a lambda related warning, which I think you are fixing in #2328 ! I'm going to give it an other try with your patch !

retif, 2020-01-18:
I managed to build Unity AirsimWrapper in MacOS

in CMakeLists.txt i was added
```
FIND_PACKAGE(Boost REQUIRED)
MESSAGE("Boost information:") 
MESSAGE("  Boost_INCLUDE_DIRS: ${Boost_INCLUDE_DIRS}") 
MESSAGE("  Boost_LIBRARIES: ${Boost_LIBRARIES}") 
MESSAGE("  Boost_LIBRARY_DIRS: ${Boost_LIBRARY_DIRS}") 
MESSAGE("  Boost_FILESYSTEM_LIBRARY: ${boost_filesystem_DIR}") 

INCLUDE_DIRECTORIES(${Boost_INCLUDE_DIRS})
LINK_DIRECTORIES("/usr/local/lib")
```
Also AirSim/Unity/build.sh contains small error:
```
if [ "$(uname)" == "Darwin" ]; then
	cp AirsimWrapper.bundle ../UnityDemo/Assets/Plugins;
else
	cp libAirsimWrapper.so ../UnityDemo/Assets/Plugins;
fi
```
AirsimWrapper.bundle is a directory so **-r** should be added to **cp**

rajat2004, 2020-01-18:
> I'm struggling with a lambda related warning, which I think you are fixing in #2328 ! I'm going to give it an other try with your patch !

Is the warning coming when building for Unreal? The one which I've fixed in the PR is in Unity, but it's one of about 350 warnings, so it's not of any importance. Still need to figure out the warnings flags.

> ```
> FIND_PACKAGE(Boost REQUIRED)
> MESSAGE("Boost information:")
> MESSAGE("  Boost_INCLUDE_DIRS: ${Boost_INCLUDE_DIRS}")
> MESSAGE("  Boost_LIBRARIES: ${Boost_LIBRARIES}")
> MESSAGE(" [...]
If possible, could you try building with the CMake changes in the PR? I don't have a Mac to test, but the build is passing so it should work. I'm not sure about hard-coding the library location, and the [manpage](https://cmake.org/cmake/help/v3.10/command/link_directories.html) also discourages it

> Also AirSim/Unity/build.sh contains small error:
> 
> ```
> if [ "$(uname)" == "Darwin" ]; then
> 	cp AirsimWrapper.bundle ../UnityDemo/Assets/Plugins;
> else
> 	cp libAirsimWrapper.so ../UnityDemo/Assets/Plugins;
> fi
> ```
> 
> AirsimWrapper.bundle is a directory so **-r** should be added to **cp**

Yes, fixed that in the Travis PR. It was also mentioned in another issue or PR of AirSim

o0Ignition0o, 2020-01-18:
> Is the warning coming when building for Unreal?
Yes exactly. I tried building from master and it worked pretty well.
> If possible, could you try building with the CMake changes in the PR ?

Sure I'm going to fetch your branch and give it a try

rajat2004, 2020-01-18:
> Edit: Ok the llvm 5.0 brew recipe seems to still be broken. I'm going to use the install script you provided on travis and try again

Well, wasn't aware of that, used on binaries on Travis since `brew install llvm` took more than 2 hrs which is the time-limit on Travis. Maybe could add the usage of binaries rather than `brew install` in `setup.sh`? Thought for another day

o0Ignition0o, 2020-01-18:
> Maybe could add the usage of binaries rather than brew install in setup.sh

Definitely ! there were also a couple of things that prevented me to run setup and build successfully ($CMAKE wasnt set in this path etc)...

I have a patch ready that seems to fix the build, but it also contains your fixes. 
Should I wait until your land your patches, or would you rather include the diff i can send you ? :)

rajat2004, 2020-01-18:
Diff would be great! I could add any additional commits required in the currently open PRs
Regarding the problem with CMake, that is included in https://github.com/microsoft/AirSim/pull/2333, the problem was that the part of code which finds the CMake binary was inside the else() function for Linux which caused the variable to not being set

retif, 2020-01-18:
While reading your PR i cant see how it can fix the problem for me, because `${Boost_LIBRARIES}` variable is empty after `FIND_PACKAGE(Boost REQUIRED)` as you can see in my previous error log:
```
Boost information:
  Boost_INCLUDE_DIRS: /usr/local/include
  Boost_LIBRARIES:
  Boost_LIBRARY_DIRS:
  Boost_FILESYSTEM_LIBRARY:
```
at my last comment i said that `LINK_DIRECTORIES("/usr/local/lib")` fixed last problems with build, normally there should be `$Boost_LIBRARY_DIRS`, but this variable is also empty as you can see.
Need to figure out why those variables stay empty.

rajat2004, 2020-01-18:
Not sure, maybe you need to specify the components such as `filesystem`?
This is the output I'm getting on Travis - 
```
CMake Warning at /usr/local/lib/cmake/boost_filesystem-1.71.0/libboost_filesystem-variant-shared.cmake:59 (message):
  Target Boost::filesystem already has an imported location
  '/usr/local/lib/libboost_filesystem-mt.dylib', which will be overwritten
  with '/usr/local/lib/libboost_filesystem.dylib'
Call Stack (most recent call first):
  /usr/local/lib/cmake/boost_filesystem-1.71.0/boost_filesystem-config.cmake:43 (include)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package)
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/Cellar/cmake/3.15.5/share/cmake/Modules/FindBoost.cmake:443 (find_package)
  CMakeLists.txt:25 (find_package)
-- Found Boost: /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components:  filesystem 
Boost information:
  Boost_INCLUDE_DIRS: /usr/local/include
  Boost_LIBRARIES: Boost::filesystem
  Boost_LIBRARY_DIRS: /usr/local/lib
  Boost_FILESYSTEM_LIBRARY: /usr/local/lib/cmake/boost_filesystem-1.71.0
```

The warning regarding the multithreaded library is something specific to Boost 1.71 which I'll test and commit the fix

o0Ignition0o, 2020-01-18:
https://gist.github.com/o0Ignition0o/0f49f17bc9077e42d92d551cfe35d700

there you go :D

Oh btw there's a lot I couldn't figure out with required variable names (C? CXX?) and there's a lot of hacky code duplication because we're not sourcing variables that we want to reuse.

All I know is that it is building on my machine.

I don't have enough build systems knowledge to know which settings make cmake happy and which don't, but I can refine it if needd be ^^'

rajat2004, 2020-01-18:
@o0Ignition0o Thanks! There's a bit of code duplication, but if it works, it works :)
CMake uses `CC` and `CXX` variables for the C and C++ compiler paths, the other variable names are just for convenience, and are passed to CMake when required. I think using the binaries is the easiest way to go, avoiding `brew` issues and also reducing the chance of messing up the system.

Also, would like to ask about the version of OSX which should be tested against in Travis. Travis doesn't have 10.15, but 10.14 is there. List can be seen here - https://docs.travis-ci.com/user/reference/osx/#macos-version
Don't think there will be much difference between versions but would like to get everyone's opinion

o0Ignition0o, 2020-01-18:
10.15 is pretty new, I would say most of the devs are either running 10.14 or 10.15, I sometimes hack on Firefox and we [need a copy of the 10.11 sdk](https://developer.mozilla.org/en-US/docs/Mozilla/Developer_guide/Build_Instructions/Mac_OS_X_Prerequisites)

10.14 will stay around for a while, i suppose it s a safe bet :)

I havent been using a mac for too long, so I would love it if someone else could give their opinion on that as well :)

Also make sure the changes I've made don't blow travis ci up (I probably removed some of the code required to make travis build pass ^^)

retif, 2020-01-19:
Next reports about running AirSim on Mac:
1. Unity Editor can launch environment with AirSim, but after stopping environment - it crashes. I Tried different versions of Unity Editor 2018 and 2019 releases, all crash after hitting environment simulation stop button. https://fogbugz.unity3d.com/default.asp?1212857_vqfelni30bmn7rna filled a bug report with their crash reporter tool.

2. I had a hard times running Unreal Editor. Recommended version UE4.18 is crashing while loading Editor(even before loading project). UE4.19 also crashing on launch. Latest version of UE4.24.1 failed to open and build AirSim project. There are compile error:
`LogCompile: Error: 'ECameraDirectorMode' does not have a 0 entry! (This is a problem when the enum is initalized by default)`
While trying different versions of UEditors i found that UE4.20.3 is able to start and successfully open, compile and run AirSim Blocks environment.

o0Ignition0o, 2020-01-19:
FWIW I had the same issue so I shifted the entries in the enum (from 0 to 7 instead of 1 to 8)

rajat2004, 2020-01-19:
> 1. Unity Editor can launch environment with AirSim, but after stopping environment - it crashes. I  Tried different versions of Unity Editor 2018 and 2019 releases, all crash after hitting [...]

Hmm, it's working on Linux for the latest Unity release (2019.2.17f1)

> 2. I had a hard times running Unreal Editor. Recommended version UE4.18 is crashing while loading Editor(even before loading project). UE4.19 also crashing on launch. Latest version of UE4.24.1 [...]

Wow, would have taken quite an effort to test all different versions! There's also a PR https://github.com/microsoft/AirSim/pull/1994 to fix a deprecation warning with 4.22
And noticing this now after a long time, there was a report of this problem in another issue https://github.com/microsoft/AirSim/issues/2348

Probably a new branch should be made for UE 4.22 or above, to get things working?

closed 2020-08-11 by madratman