microsoft/AirSim #2377: Lidar acts weirdly

ivannson, 2020-01-13:
I am running Airsim in Unreal 4.22.3, on Ubuntu 18.04 server. I am running Unreal by using a virtual display, along with virtual gl and Turbo VNC. 

Below are my Lidar settings:
```
"LidarSensor1": {
					"SensorType": 6,
					"Enabled" : true,
					"NumberOfChannels": 32,
					"Range":100,
					"RotationsPerSecond": 5,
					"PointsPerSecond": 40000,
					"X": 0, "Y": 0, "Z": -1.73,
					"Roll": 0, "Pitch": 0, "Yaw" : 0,
					"VerticalFOVUpper": 3,
					"VerticalFOVLower": -25,
					"HorizontalFOVStart": -180,
					"HorizontalFOVEnd": 180,
					"DrawDebugPoints": false,
					"DataFrame": "SensorLocalFrame"
```

However, when I run Lidar data collection, the number of points in each scan is around 5000, not 40000 as specified. 

In rare occasions when I am getting around 35k points, the performance drops to 1 FPS, and stays there unless I restart Unreal. Disabling Lidar doesn't work, still stays at 1 FPS. What's weird, it's not only the simulation running very slowly, but the entire UE4 editor is running at 1 FPS. 

I am also attaching a screenshot showing the output of `vglrun glxgears` and starting a simulation when it gets the right number of Lidar points.
[image]

ironclownfish, 2020-01-21:
Hi @ivannson 
Lidar doesn't return full scans, but successive chunks of a full scan. It's sort of like real-world lidar, which streams out points as they're scanned, rather than as the entire cloud at once. If you aggregate the points you're receiving, you should see that a full sweep is completed every 0.2 seconds, given your settings.

ivannson, 2020-01-24:
Yes, I think the issue was that the lidar rotation speed was not able to keep up with the frame rate. When running at 50 fps, and lidar rotating at 10 rotations per second, it was only able to capture the scene in that frame, and carried on in the next frame.

I solved it by setting a fixed frame rate in UE to 20 fps, and putting lidar rotation speed to 20 rotations per second.

closed 2020-01-24 by ivannson