microsoft/AirSim #2378: connect between Airsim on Window10 and PX4 on Ubuntu

xogns8121, 2020-01-14:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

I want to connect between Airsim on Window10 and PX4 on Ubuntu.
I found all over how to connect between Airsim on Window 10 and PX4 on Ubuntu but there no idea.

1. 
[image]

Im waiting like this on Ubuntu.

2. 
I have no idea which ip, port to change so i changed all of ip and port.

[image]
[image]

What should i change? also is it possible connect between 2 PCs?

rajat2004, 2020-01-16:
It should be possible to form the connection between 2 different machines; default settings are general for running everything on a single system.
The most likely culprit here is that the IP addresses specified are incorrect, the `LocalHostIp` should be the IP of this machine, i.e. the IP address of the machine running AirSim for the network adapter that you want to use, such as ethernet or WiFi. Different network adapters will have different IP, such as `127.0.0.1` which is the default for LocalHost
The `UdpIp` needs to be set to the machine running PX4
You have also specified the Serial settings which is wrong since it will then not use UDP and instead look for a Pixhawk or any other Flight controller connected to the computer using USB

As a side note, it's better to paste the settings as text in a code block for easier reading.

xogns8121, 2020-01-20:
> It should be possible to form the connection between 2 different machines; default settings are general for running everything on a single system.
> The most likely culprit here is that the IP [...]

Thank you for reply
It is helpful that use `UdpIp` for running PX4.

And then is it possible linking multi PX4 sitl with airsim??

lovettchris, 2021-06-04:
Being able to connect to a remote Linux machine on a different IP address is now supported.  It was added with this fix: 
https://github.com/microsoft/AirSim/pull/3603.  Please take note of the new PX4 feature used there which supports PX4_SIM_HOST_ADDR.  This requires PX4 latest bits.