microsoft/AirSim #2379: Customising Python API for FPV camera FOV control - (+simSetCameraFov function)

ibrahim0v, 2020-01-14:
Hello @madratman I hope and think that you'll be the one to help me :)

I am currently working on a project where I need an controllable gimbal camera but with the additive of zoom function. I found out that I can zoom in and out of a view by altering field of view parameter of the camera. Therefore I have already done that in airSim. Now in my version, when FPV mode is entered with "F" key, the camera orientation can be controlled with WASDQE keys and its fov can be controlled with the arrow keys. Anyways, now I want to be able to do this using the API you provided. The camera orientation can already be configured using the **simSetCameraOrientation**.  
So I started working on declaring and using a function called **simSetCameraFov** that function as its name implies but couldn't reach a point yet. So please your help would be appreciated. I will be working on it and I would like to post a merge request for an API update + the manual fpv gimbal control so the community can get use of it.

ibrahim0v, 2020-01-15:
I am able to change FOV of the camera now using my simSetCameraFov function but something is updating the FOV back to settings.fov which is 90 degress by default. I cannot find the module that is reverting it back to 90deg. 

@clovett @sytelus @madratman Can someone please guide me

ibrahim0v, 2020-01-16:
I have solved it. Now I have successfully added simSetCameraFov function as api for FPV camera zoom control. I can post a pull request if anyone cares

saihv, 2020-01-21:
A pull request would be very welcome!

ibrahim0v, 2020-01-21:
@saihv okay I'll post a PR as soon as I finish up some cleaning in my code :+1:

alwynmathew, 2020-01-30:
@idanlevin what is the FOV you achieved? Will it be able to modify your setting to 180 degree fisheye camera?

ibrahim0v, 2020-02-07:
I have limited it between 150degs to 20degs but the limits are customisable. I have seen 180 and the upside down view for degs above 180 (upto 360 I guess)

alwynmathew, 2020-02-07:
@ibrahim0v Have you posted this PR yet? I need 180 degree Horizontal FOV, how would I do it? Any help would be appreciated.

ibrahim0v, 2020-02-07:
I'll post a PR ASAP :+1:

rajat2004, 2020-04-14:
https://github.com/microsoft/AirSim/pull/2534 has been merged which closes this

ibrahim0v, 2020-05-12:
thanks @rajat2004 I couldn't send my PR :) Yours is a better way anyways :+1:

closed 2020-05-12 by ibrahim0v