microsoft/AirSim #2385: Lidar offset in the 'Setting.h doesn't' mean the 'relative' position to the drone body frame

didwjdrl12, 2020-01-16:
Hi, I'm Jeonggi Yang in Republic of Korea. 
We developed some simulators by using AirSim. Thanks you.
Recently, we found some bugs of Lidar offset function.

As you mention, 

X Y Z | Position of the lidar relative to the vehicle (in NED, in   meters)

in settings.h file.

But the lidar position is not moved when I set the X:5 and rotated the drone in the yaw directions

[image]
[image]

And in the code, Ray starting point is not affected with Drone's attitude.
Please, fix this phenomenon.

Thank you.

rajat2004, 2020-01-16:
Hi, is this the behaviour with the latest master? https://github.com/microsoft/AirSim/pull/2143 has been merged which I think fixed this problem

didwjdrl12, 2020-01-17:
Hi,
We developed some projects on old version. We didn't know that It was solved lately.
I think It is okay to close the issue.
 
Thank you.

closed 2020-01-17 by saihv