microsoft/AirSim #2386: How to connect client on Ubuntu to a car simulation on Windows

ivannson, 2020-01-17:
Hello,

I've seen docs/issues talking about running a simulation on Windows and running Python API on a Linux machine. 

Is it possible to do the same for a car simulation? I haven't found much information in the docs on how to set it up. So far I have tried the following:

```Python
self.client = airsim.CarClient(ip="xx.xx.xxx.xx")
self.client.confirmConnection()
```

However I am unable to confirm the connection (`WARNING:tornado.general:Connect error on fd 9: ETIMEDOUT`).

I have attempted turning off Firewall on the Windows PC.

A few questions relating to the issue:
- Is the Python client connecting to the server using TCP?
- When simulation starts in Unreal, is the 41451 port open automatically?

ivannson, 2020-01-29:
Is this a built in functionality? If not then I will have to think of a work around, for example manually forwarding ports from Ubuntu to Windows.

jonyMarino, 2020-08-14:
@ivannson were you able to run this?

jonyMarino, 2020-08-14:
duplicated on #2422

closed 2020-08-14 by jonyMarino

ivannson, 2020-08-15:
@jonyMarino I was able to run it but with 2 Windows machines (not between Windows and Linux), but that may have had something to do with the firewall settings on the Linux server, couldn't even ping it.

jonyMarino, 2020-08-17:
@ivannson that is great. Please reopen if you can ping between them and not able to run it.