microsoft/AirSim #2387: Speed not matching with the timestamps

Pablitinho, 2020-01-17:
Hi People,

I am using v1.2.0 - Windows (10) Neighbourhood (small urban neighbourhood block in Car mode), 
with a Nvidia Quadro M2000M

My setting file is:
```
{

  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "ClockSpeed": 1.0,
  "SpeedUnitFactor": 1.0,
  "SpeedUnitLabel": "m/s",
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.033,
    "Cameras": [
        { "CameraName": "0", "ImageType": 0, "PixelsAsFloat": false, "Compress": true },
		{ "CameraName": "0", "ImageType": 2, "PixelsAsFloat": true, "Compress": false }
    ]
  },
	
  "CameraDefaults": {
    "CaptureSettings": [
	  {
        "ImageType": 0,
        "Width": 1280,
        "Height": 800,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 30,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.44,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12 
	  },
      {
        "ImageType": 1,
        "Width": 1280,
        "Height": 800,
        "FOV_Degrees": 90,         
        "OrthoWidth": 5.12 
	  },
      {
        "ImageType": 2,
        "Width": 1280,
        "Height": 800,
        "FOV_Degrees": 90,         
        "OrthoWidth": 5.12 
	  },
	  {
        "ImageType": 3,
        "Width": 1280,
        "Height": 800,
        "FOV_Degrees": 90,         
        "OrthoWidth": 5.12 
	  },
	  {
        "ImageType": 7,
        "Width": 1280,
        "Height": 800,
        "FOV_Degrees": 90,         
        "OrthoWidth": 5.12 
	  }
	  
    ]
	}
}
```
I am driving straight ahead until the fence and I am getting these values in the airsim_rec.txt:
```
......
1579273490734	100.481209	-0.013188	-0.595794	0.999999	0.000020	0.001526	0.000223	1.000000	0.000000	0.000000	4	0	3807.770508	21.534914	img_0_0_1579273490664268800.png;img_0_2_1579273490672230200.pfm
1579273491227	104.819077	-0.011261	-0.600899	0.999995	0.000019	0.003159	0.000247	1.000000	0.000000	0.000000	4	0	3790.055664	21.873173	img_0_0_1579273491153727800.png;img_0_2_1579273491161648800.pfm
1579273491747	109.219688	-0.008708	-0.607839	0.999998	-0.000360	0.002006	0.000263	1.000000	0.000000	0.000000	4	0	3783.358154	22.200687	img_0_0_1579273491673979600.png;img_0_2_1579273491680617800.pfm
1579273492242	113.686241	-0.006978	-0.606595	0.999999	-0.000232	0.001429	0.000325	1.000000	0.000000	0.000000	4	0	3790.760742	22.523422	img_0_0_1579273492172626300.png;img_0_2_1579273492178518500.pfm
1579273492737	118.215485	-0.003929	-0.607300	0.999999	-0.001073	0.000553	0.000315	1.000000	0.000000	0.000000	4	0	3817.269531	22.833920	img_0_0_1579273492665380800.png;img_0_2_1579273492671869900.pfm
1579273493206	122.809967	-0.002637	-0.607990	0.999998	-0.000174	0.002114	0.000361	1.000000	0.000000	0.000000	4	0	3847.889893	23.140617	img_0_0_1579273493135205900.png;img_0_2_1579273493141911400.pfm
1579273493703	127.462006	0.001133	-0.607102	0.999999	-0.000642	0.000994	0.000356	1.000000	0.000000	0.000000	4	0	3880.597168	23.444881	img_0_0_1579273493633699300.png;img_0_2_1579273493640118200.pfm

```
 dTime= (1579273493206-1579273492737)/1000 = 0.469 s
 dDisplacement = 122.809967-118.215485 = 4.594482 m

Speed = dDisplacement/dTime = 4.594482/0.469 =9.79 m/s

But I am getting 23.140617 m/s. So.... there is something wrong.....

After some tests, I was running in another computer (with a Nvidia 2080 ) and the DTime was lower and the speed was matching properly. 

That's mean that the kind of GPU affect to the results, but in theory it should not. Why?
Can I modify something in my settings file to avoid this problem? 

Thanks in advance guys!!!!!

Ckq-Sugar, 2022-10-20:
Hi, Pablitinho
I have a similar problem too. Set the camera interval to 0.033 (RecordInterval: 0.033) but the timestamps of the saved images cannot meet this requirement (the time interval is about 0.2 seconds)
Have you solved the problem now?

Pablitinho, 2022-10-20:
Hi,
Time ago I was using different GPU and I was getting different results. Give a try with a different machine. I don’t know your setup…