microsoft/AirSim #2388: drone_image_raw.py ros image publishing

kwads123, 2020-01-17:
Hi all. I'm hoping someone can help me. I am trying to publish images from AirSim with ROS using the drone_image_raw.py script. It appears to run correctly and publishes data verified by rostopic echo ...

The problem is with the image data. I have received data in MATLAB but the data part of the image message says that it is a 110529 x 1 uint8 string. I haven't changed anything in the python script or in the settings file. In fact, I am trying this with the default settings.json file. Does anyone know the image format or how to process this data in MATLAB to get it into an RGB format?

marufino, 2020-01-31:
Take a look at the Python example from the image_apis page https://github.com/microsoft/AirSim/blob/master/docs/image_apis.md#using-airsim-images-with-numpy
Should be pretty similar since numpy syntax and matlab are close

jonyMarino, 2020-08-17:
@kwads123 could you solve your problem? If so, could you post it in this issue?