microsoft/AirSim #2391: Is there AirSim support for a Skid Steering vehicle, like a rover or tank? If not it would be a cool feature to include

Ishira1601, 2020-01-20:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

mitchellspryn, 2020-01-27:
You could try [this fork]. It's based off an older version of AirSim, but should be able to do a skid steer bot (that's what I'm currently using it for).

Ishira1601, 2020-02-02:
It doesn't include any skid steer vehicles examples at the moment i suppose, to get a good starting point.

Ishira1601, 2020-02-02:
> You could try [this fork](https://github.com/mitchellspryn/UrdfSim). It's based off an older version of AirSim, but should be able to do a skid steer bot (that's what I'm currently using it for).
It doesn't include any skid steer vehicles examples at the moment i suppose, to get a good starting point.

mitchellspryn, 2020-02-02:
Check out the Lunabot [here](https://github.com/mitchellspryn/UrdfSim/tree/master/Examples/UrdfBot). That's a skid steer bot.