microsoft/AirSim #2392: connect between px4 ubuntu and airsim window 10

xogns8121, 2020-01-21:
First, is it possible connect between external ip with px4 and airsim?
I connected px4 and airsim on local computer ubuntu. and then i m trying to connect between external computer. 
What should i change settings?
1. Airsim : I changed airsim settings like this

[image]
[image]

2. PX4 : Waiting like this but nothing changed and nothing connected.
 What should i change at px4?

[image]

tahameg, 2020-01-21:
I am also working on this and will let you know when I am done

xogns8121, 2020-01-22:
> I am also working on this and will let you know when I am done

thank you

tahameg, 2020-02-24:
I have tested the following method but it results in too many Gyro and Accel fail errors. I am not sure if the connection is successfully achieved between the sitl and airsim. But it a kind of works. This method constructs an ssh bridge between port 4560 of sitl device and same port of simulation device. 

first set up and OpenSSH server on Windows 10. You can find many resources for achieving this. After activating  the ssh server, open a console on sitl device and type

`ssh -L 4560:127.0.0.1:4560 <user-name>@<local-ip-of-simulation-device>`

Then type your password. After you successfully complete this step, start the sitl when airsim is working. Let me know if you find any mistakes on this method.

xogns8121, 2020-02-25:
Ok i'll try. Thank you so.
but, did you try to sitl connection on single PC??
I also get so many Gyro and Accel fail errors.  Now, I'm trying to fix this error. and then i will try to  ssh connection.
(That is odd that when i connected sitl on single PC a month ago, there were no error at all. )

tahameg, 2020-02-25:
Yess, it turns out that this error arises from latency related issues. I guess i was getting it because data transmission speed was not fast enough, port forwarding and transmission over wifi causes extensive overhead. In your case, errors are probably happening because your machine is not powerful enough to run a simulated linux console and simulation itself together. I suggest you to try the method i suggested above with an cable connection.

xogns8121, 2020-02-26:
Okay i understand, I'll try it. thank you

also I have a question tahameg.
I knew that there are several method to connect remote PC.
and you chose ssh method.
I tried < changing setting.json >, <mavlink router> methods.
but method <changing setting.json > is not work properly.
I changed UdpIp to remote px4 machine but it doesn't work properly.
did you also get this error?? is there any error??

xogns8121, 2020-02-27:
@tahameg  
px4 Firmware 1.10.x, there is continuous gyro and accel error so i can't arming or moving
but px4 Firmware 1.9.x there is not continuous just intermittent error. additionally, i can arming or controlling drone.

tahameg, 2020-02-27:
> Okay i understand, I'll try it. thank you
> 
> also I have a question tahameg.
> I knew that there are several method to connect remote PC.
> and you chose ssh method.
> I tried < changing setting.json >, methods.
> but method is not work properly.
> I changed UdpIp to remote px4 machine but it doesn't work properly.
> did you also get this error?? is there any error??
> 
> I tried ssh connection but there is still gyro and accel error again and again.......

I dont know what is UdpIp parameter is actually doing, it may be depricated. Connection between sitl and airsim is performed through tcp port 4560 not any Udp port. So, i guess changing UdpIp parameter will not give the result. At least I couldn't make it work.

As for the errors, as i said they are probably because of latency. You should directly connect 2 devices with an ethernet cable. Then, set up ip adresses(I can help about this further if you can't find how to do it). Start ssh tunneling with the cable connection. This will minimize latency, if you still get the errors, it is probably because your device is unable to run the simulation with target fps rate.

xogns8121, 2020-02-28:
I understand. thank you so much.
Anyway i connected remote PC with your help.
I will test more. and if it works well without any error i'll tell you. 

also did you finish your doing??

tahameg, 2020-02-28:
> I understand. thank you so much.
> Anyway i connected remote PC with your help.
> I will test more. and if it works well without any error i'll tell you.
> 
> also did you finish your doing??

It is nice to hear that pal 👍 Please let me know if you find a better way

lovettchris, 2021-06-04:
Some of these gyro latency errors will be fixed with the latest AirSIm bits where LockStep was fixed by using SteppableClock which can now be specified in your [settings.json](https://microsoft.github.io/AirSim/px4_sitl/).  With these fixes PX4 will not see any latency because it is getting a fake timestamp generated by AirSim.

closed 2022-05-30 by jonyMarino