microsoft/AirSim #2394: PX4 HITL + AirSim: No GPS Lock problem

ibrahim0v, 2020-01-22:
I have been testing my algorithms using the v.1.2.2 tag branch until today. After seeing mr @clovett 's commits (12th Dec) I have decided to checkout to master to see if they solve some of my problems (Repeating Critical Navigation Failure and sensor data losses, freezing when AirSim is close and etc.). But unfortunately now AirSim neither does send GPS data nor it gets actuators from my Pixhawk2Cube. I think there is a problem with my old AirSim settings(.json) as the Pixhawk settings are now updated (params are added and all).

I am running on the following system:
Pixhawk2Cube + v1.9.2 fmu
Pixhawk and AirSim are connected through serial port (`use_serial=tru`e)
QGroundControl communicating to my Pixhawk through `telem1` port (another direct mavlink instance, instead of the AirSim udp port)
A joystick with `RC_IN_MODE `= Virtual RC by Joystick

My settings are as follows:
```
    "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor",
      "Model": "Kargu",
      "QgcHostIp": "127.0.0.1",
      "QgcPort": 14550,
      "ControlIp": "127.0.0.1",
      "ControlPort": 14560,
      "SerialBaudRate": 115200,
      "SerialPort": "*",
      "SimCompID": 42,
      "SimSysID": 142,
      "UseSerial": true,
      "UseTcp": false,
      "VehicleCompID": 1,
      "VehicleSysID": 142,
      "LocalHostIp": "127.0.0.1",
      "params": {
        "NAV_RCL_ACT": 0,
        "NAV_DLL_ACT": 0,
        "LPE_LAT": 47.641468,
        "LPE_LON": -122.140165,
        "COM_OBL_ACT": 1
                }
    }
}
}
```

ibrahim0v, 2020-01-22:
I have checked if the problem can be reproduced using sitl. The problem is consistent only in the hitl case and gps is successfully sent from AirSim to px4_sitl

lovettchris, 2020-01-23:
Did you build this version of AirSim and the Unreal environment you are using yourself or are you using pre-canned binaries from someplace?

ibrahim0v, 2020-01-23:
I've built it them by my self on Ubuntu 18.04. I think I am misusing the settings but I could not figure out it yet

ibrahim0v, 2020-01-23:
I solved it!

ibrahim0v, 2020-01-23:
I have partially figured out the problem, I think the failure sensor data pass to HITL is caused by one of the flags in MavLinkMultirotorApi: 98 -> `if (sensors_ == nullptr || !connected_ || connection_ == nullptr || !connection_->isOpen() || !got_first_heartbeat_)`. This is applicable in SITL but not in HITL. As if-notting results a successful pass of sensor data but no actuators yet.

lovettchris, 2020-01-24:
I accidentally broke the PX4 serial connection when I did the lockstep fixes.  My very sincere apologies for the time you've had to waste on this, I thought I tested serial, but not well enough.   

Can you try this fix: https://github.com/microsoft/AirSim/pull/2397

ibrahim0v, 2020-01-27:
Mr @clovett , thanks! Now the connection is okay but the sensors problem is insisting. The GPS and sensor data are not being sent yet. I have ammended the 98th line in MavLinkMultirotorApi.hpp as             `if (**(**sensors_ == nullptr || !connected_ || connection_ == nullptr || !connection_->isOpen() || !got_first_heartbeat_**) && 0**)`
`                return;`
as a workaround to this and now I can make run my tests with no problems and better than before thanks to your fixes :+1:

closed 2020-01-28 by ibrahim0v