microsoft/AirSim #2396: Camera Calibration Values

BrianNguyen214, 2020-01-24:
I was wondering if anyone had the camera calibration values for the camera used in AirSim. Currently, I'm interested in using AirSim as a tool to test out SLAM and object detection algorithms, and in order to do so, I need the camera calibration values for the camera. 

If no one does, then I'm wondering if anyone can help me create the checkerboard in AirSim, which is required for the camera calibration procedure.

saihv, 2020-01-25:
You can approximate the camera intrinsics using the following formula

Horizontal FoV = 2 * arctan( width / 2f ) (then assuming fx = fy = f)

The default field of view of the camera is 90 degrees, so your focal length ends up being half of the image width in pixels. cx and cy can be set to w/2 and h/2 as well.

See the discussion in https://github.com/microsoft/AirSim/issues/269 for more details.

closed 2020-01-25 by saihv

NickPerezCarletonUniversity, 2021-10-01:
@saihv what would be the distortion parameters? I imagine they would all be set to 0. The experimentation done here: https://github.com/microsoft/AirSim/issues/269#issuecomment-313526623 indicates this.

See here, the distortion parameters, k1, k2, k3, p1, p2: https://docs.opencv.org/4.5.3/dc/dbb/tutorial_py_calibration.html 
I noticed orb slam does not use k3. the experimentation I referenced also only shows 4 distortion parameters.

saihv, 2021-11-22:
@NickPerezCarletonUniversity By default, yes, the distortion parameters are all zeroes. You can change them to desired values through our [camera distortion API](https://microsoft.github.io/AirSim/image_apis/#camera-apis)

cumtchenchang, 2022-01-05:
Can Airsim support fisheye model, such as Kananla Brandt 8 model?
Thank you

emeQee, 2022-02-13:
@saihv Is the sensor size know or pixel size in mm? Or is there a way to get the intrinsic matrix in SI units?

hkumarcodes, 2023-12-22:
Is this camera calibration valid for other FOV's? I cannot seem to get good results using the above formula.