microsoft/AirSim #2401: How many unreal units are equal to 1 airsim meter?

tahameg, 2020-01-30:
how many units should I move an object to match 1 meter in simulation space?

WouterJansen, 2020-02-01:
AirSim uses meters, so all configurations and data coming from the API will be in meters as far as I know.

Unreal uses centimeters. So moving an object in the editor for 100 units will have moved it by 1 meter.

tahameg, 2020-02-01:
> AirSim uses meters, so all configurations and data coming from the API will be in meters as far as I know.
> 
> Unreal uses centimeters. So moving an object in the editor for 100 units will have moved it by 1 meter.

Thank you so much 👍🏻

closed 2020-02-01 by tahameg

krishnarajk, 2022-03-22:
So will it go exactly 100 units in unreal if we give 1 as an argument in Arisim move API? It is not moving exactly 100 units for me.

WouterJansen, 2022-03-22:
which move API call? AFAIK they all use NED and radians/meters.

krishnarajk, 2022-03-22:
MoveToPositionAsync(). I want to move the drone in unreal engine 1 meter on the x axis. When I give 1 as x value, it moves more than 100 unreal units

krishnarajk, 2022-03-22:
Okay. But if my starting position is 0,0,0 and I want to move one meter on the x-axis then the x,y,z will be 1,0,0. Is this the correct approach?

WouterJansen, 2022-03-22:
That is indeed what this command should do.

krishnarajk, 2022-03-23:
[image]

Will this be a reason for moving more than 100uu in unreal engine if I try to move 1m. In unreal engine it is showing that 1m ~ 125uu.

WouterJansen, 2022-03-24:
If with uu you mean unreal-unit, the units used in unreal are cm. So a meter in unreal is 100uu/100cm.

krishnarajk, 2022-03-24:
But do you have any idea why it is showing in the above image, 1m in more than 100uu (there are approx 125uu).

WouterJansen, 2022-03-24:
I don't know. Where does that scale come from? You cannot change unreal engine in that way. 1m=100cm=100 units in unreal

krishnarajk, 2022-03-24:
[image]

When I changed the view perspective I got this scale.

LukeAllen, 2023-02-14:
For completeness, note that the Z axis is also flipped between the two systems (AirSim uses +Z = down, because they wanted a conventional right-hand coordinate system). So the full conversion from default Unreal editor units to Airsim units is:
```
airsim_x = .01*unreal_x
airsim_y = .01*unreal_y
airsim_z = -.01*unreal_z
```

The docs mention it here: https://microsoft.github.io/AirSim/apis/#coordinate-system