microsoft/AirSim #2404: ROS wrapper doesn't work properly with PX4

sourishg, 2020-02-03:
I'm trying to use the ROS wrapper provided with AirSim along with a PX4 quadrotor. However, I get an error when I try to launch `airsim_node.launch`:
```
Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
Exception raised by the API, something went wrong.
rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Cannot perform operation when no vehicle is connected or vehicle is not responding.
AirsimROSWrapper Initialized!
```
Although most topics seem to be published fine (e.g., lidar topic etc.), the odometry topic (`/airsim_node/PX4/odom_local_ned`) is faulty, i.e., it always publishes (0, 0, 0) as position. My settings file looks like this:
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockSpeed": 1.0,
  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "UseTcp": true,
      "TcpPort": 4560,
      "ControlPort": 14510,
      "params": {
        "NAV_RCL_ACT": 0,
        "NAV_DLL_ACT": 0,
        "LPE_LAT": 47.641468,
        "LPE_LON": -122.140165,
        "COM_OBL_ACT": 1
      },
      "Sensors": {
        "Barometer": {
          "SensorType": 1,
          "Enabled": true
        },
        "Imu": {
           "SensorType": 2,
           "Enabled": true
        },
        "Gps": {
           "SensorType": 3,
           "Enabled": true
        },
        "Magnetometer": {
           "SensorType": 4,
           "Enabled": true
        },
        "Distance": {
           "SensorType": 5,
           "Enabled": true
        },
        "LidarCustom": {
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 16,
          "PointsPerSecond": 10000,
          "X": 0,
          "Y": 0,
          "Z": -1,
          "DrawDebugPoints": true
        }
      },
      "Cameras": {
        "front_center_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 27,
              "DepthOfFieldFstop": 2.8,
              "DepthOfFieldFocalDistance": 200.0, 
              "DepthOfFieldFocalRegion": 200.0,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": 0, "Z": 0.10,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        },
        "front_left_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }, 
            {
              "PublishToRos": 1,
              "ImageType": 1,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": -0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "front_right_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": 0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 1, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  },
  "SubWindows": [
    {"WindowID": 0, "ImageType": 0, "CameraName": "front_left_custom", "Visible": true},
    {"WindowID": 1, "ImageType": 0, "CameraName": "front_center_custom", "Visible": false},
    {"WindowID": 2, "ImageType": 0, "CameraName": "front_right_custom", "Visible": true}
  ]
}
```
Any help on how to fix this will be appreciated! Note that I have to change the `ControlPort` to 14510 in order to let mavros connect with SITL (using port 14580). However, this change doesn't affect anything.

The odometry topic is fine when using SimpleFlight but with PX4 it doesn't work.

Zartris, 2020-04-27:
Hi Sourishg 
Did you manage to fix this??

sourishg, 2020-04-27:
Hi @Zartris, unfortunately no. It seems AirSim's API control (`enableApiControl`) won't work with PX4 which kind of makes sense because AirSim internally uses SimpleFlight for its control API. So you'd technically use PX4 for control and use this wrapper for getting state/sensor information.

A workaround for this problem would be to modify the existing code to turn off API control. I switched to Python ROS wrapper for AirSim so unfortunately, I didn't try it out. You can try to turn off the API control and see if it works (in theory it should).

deunsoljo, 2020-09-25:
I think this problem happened when the drone cannot receive starting point data (i.e. x=0,y=0,z=0) from the PX4.

I can't answer correctly because I'm not currently doing these experiments, but in the past, when I was working on it, to solve this problem, 
First, run the Airsim simulator  
Second, connect the PX4 based on SILS or HILS to receive corresponding data
Third, run the airsim_node.launch. (it must be conducted sequentially)

When I conducted this sequence(First ~ Third), I was able to verify that the airsim_node.launch process worked well without exception.

closed 2021-07-23 by jonyMarino

jonyMarino, 2021-12-15:
I made it work using these settings:
```
{ 
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles": {
        "PX4": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "QgcHostIp": "",
            "TcpPort": 4560,
            "ControlIp": "remote",
            "ControlPort": 14580,
            "LocalHostIp": "172.22.224.1",            
            "Sensors": {
                "barometer": {
                    "SensorType": 1,
                    "Enabled": true,
                    "pressure_factor_sigma": 0.0001825
                },
                "Imu" : {
                    "SensorType": 2,
                    "Enabled": true
                },
                "LidarCustom": {
                    "SensorType": 6,
                    "Enabled": true,
                    "NumberOfChannels": 16,
                    "PointsPerSecond": 10000,
                    "X": 0,
                    "Y": 0,
                    "Z": -1,
                    "DrawDebugPoints": true
                }
            },
            "Cameras": {
                "front_center_custom": {
                    "CaptureSettings": [
                    {
                        "PublishToRos": 1,
                        "ImageType": 0,
                        "Width": 640,
                        "Height": 480,
                        "FOV_Degrees": 27,
                        "DepthOfFieldFstop": 2.8,
                        "DepthOfFieldFocalDistance": 200.0, 
                        "DepthOfFieldFocalRegion": 200.0,
                        "TargetGamma": 1.5
                    }
                    ],
                    "X": 0.50, "Y": 0, "Z": 0.10,
                    "Pitch": 0, "Roll": 0, "Yaw": 0
                },
                "front_left_custom": {
                    "CaptureSettings": [
                    {
                        "PublishToRos": 1,
                        "ImageType": 0,
                        "Width": 672,
                        "Height": 376,
                        "FOV_Degrees": 90,
                        "TargetGamma": 1.5
                    }, 
                    {
                        "PublishToRos": 1,
                        "ImageType": 1,
                        "Width": 672,
                        "Height": 376,
                        "FOV_Degrees": 90,
                        "TargetGamma": 1.5
                    }
                    ],
                    "X": 0.50, "Y": -0.06, "Z": 0.10,
                    "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
                },
                "front_right_custom": {
                    "CaptureSettings": [
                    {
                        "PublishToRos": 1,
                        "ImageType": 0,
                        "Width": 672,
                        "Height": 376,
                        "FOV_Degrees": 90,
                        "TargetGamma": 1.5
                    }
                    ],
                    "X": 0.50, "Y": 0.06, "Z": 0.10,
                    "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
                }
            },
            "Parameters": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "COM_OBL_ACT": 1,
                "LPE_LAT": 47.641468,
                "LPE_LON": -122.140165
            },
            "Logs": "d:\\temp\\mavlink"
        }
    }
}
```