microsoft/AirSim #2405: UE4 freezes when I hit ESC button during px4 simulation

icsl-Jeon, 2020-02-05:
Hello.

I am struggling with 'stop' and 'play' button in UE4 under AirsimGameMode. 

First of all, if I spawn a SimpleFlight Drone on a world, I can easily play and stop with ESC button. 

However, when I play with px4 vehicle, UE4 freezes while trying to disconnt mavlink seemingly. 

Any help on this? 

====

*px4 side*
```
SITL ARGS
sitl_bin: /home/jbs/lib/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: none
model: iris
src_path: /home/jbs/lib/Firmware
build_path: /home/jbs/lib/Firmware/build/px4_sitl_default
SITL COMMAND: "/home/jbs/lib/Firmware/build/px4_sitl_default/bin/px4" "/home/jbs/lib/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/jbs/lib/Firmware"/test_data
INFO  [px4] Creating symlink /home/jbs/lib/Firmware/ROMFS/px4fmu_common -> /home/jbs/lib/Firmware/build/px4_sitl_default/tmp/rootfs/etc
0 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2020-02-05/06_42_02.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [mavlink] partner IP: 127.0.0.1

```

*Unreal side* 
```
[2020.02.05-06.42.01:062][ 50]LogTemp: Disconnecting mavlink vehicle
[2020.02.05-06.42.01:062][ 50]LogTemp: Waiting for TCP connection on port 4560, local IP 127.0.0.1
[2020.02.05-06.42.01:062][ 50]LogTemp: Connected to SITL over TCP.
[2020.02.05-06.42.01:062][ 50]LogTemp: received first heartbeat
[2020.02.05-06.42.01:062][ 50]LogTemp: Connecting to PX4 Control UDP port 14580, local IP 127.0.0.1, remote IP...
[2020.02.05-06.42.01:062][ 50]LogScriptCore: Warning: Script Msg: Attempted to access index 0 from array 'CallFunc_GetAllActorsOfClass_OutActors' of length 0 in '/AirSim/Weather/WeatherFX/WeatherActor.WeatherActor_C:ExecuteUbergraph_WeatherActor'!
[2020.02.05-06.42.01:344][ 50]LogTemp: PX4
PX4
[2020.02.05-06.42.01:348][ 50]PIE: Play in editor start time for /Game/ThirdPersonBP/Maps/UEDPIE_0_ThirdPersonExampleMap 57.348
[2020.02.05-06.42.01:349][ 50]LogBlueprintUserMessages: Late PlayInEditor Detection: Level '/Game/ThirdPersonBP/Maps/ThirdPersonExampleMap.ThirdPersonExampleMap:PersistentLevel' has LevelScriptBlueprint '/Game/ThirdPersonBP/Maps/ThirdPersonExampleMap.ThirdPersonExampleMap:PersistentLevel.ThirdPersonExampleMap' with GeneratedClass '/Game/ThirdPersonBP/Maps/ThirdPersonExampleMap.ThirdPersonExampleMap_C' with ClassGeneratedBy '/Game/ThirdPersonBP/Maps/ThirdPersonExampleMap.ThirdPersonExampleMap:PersistentLevel.ThirdPersonExampleMap'
Ground control connected over UDP.
[2020.02.05-06.42.03:005][111]LogTemp: Ground control connected over UDP.
Enabling lockstep mode
[2020.02.05-06.42.03:034][112]LogTemp: Enabling lockstep mode
[logger] file: ./log/2020-02-05/06_42_02.ulg
[2020.02.05-06.42.03:306][122]LogTemp: [logger] file: ./log/2020-02-05/06_42_02.ulg
[2020.02.05-06.42.03:695][136]LogBlueprintUserMessages: Early EndPlayMap Detection: Level '/Game/ThirdPersonBP/Maps/ThirdPersonExampleMap.ThirdPersonExampleMap:PersistentLevel' has LevelScriptBlueprint '/Game/ThirdPersonBP/Maps/ThirdPersonExampleMap.ThirdPersonExampleMap:PersistentLevel.ThirdPersonExampleMap' with GeneratedClass '/Game/ThirdPersonBP/Maps/ThirdPersonExampleMap.ThirdPersonExampleMap_C' with ClassGeneratedBy '/Game/ThirdPersonBP/Maps/ThirdPersonExampleMap.ThirdPersonExampleMap:PersistentLevel.ThirdPersonExampleMap'
[2020.02.05-06.42.03:697][136]LogTemp: Disconnecting mavlink vehicle
Disconnecting mavlink vehicle

```

ibrahim0v, 2020-02-12:
Hey, @clovett solved this in #2397

depauwjimmy, 2020-03-02:
Trying this today with the last master code but it is still happening.

jonyMarino, 2020-08-14:
@depauwjimmy and @icsl-Jeon Please feel free to open a new issue if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

bwassim, 2020-08-27:
The issue still presist. 
I am on the branch v1.3.1-linux for AirSim / Ubuntu 18.04. Without even launching px4, just the fact doing a Play and then Stop, freezes the Unreal Editor in either block or LandscapeMountain environment. 
if we launch the "make px4_sitl_default none_iris" and then the simulation, the fact of using "shutdown" command in terminal will generate the following error you see on the UE. Why "hil" is appearing here? may be that's the source of the error? since SITL is used
{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles": {
        "PX4": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14580,
            "params": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "LPE_LAT": 47.641468,
                "LPE_LON": -122.140165,
                "COM_OBL_ACT": 1
            }
        }
    }
}
[image]

lovettchris, 2020-08-28:
the name "hil" is not an issue, it is just the internal name we use on the connection object, it is the same name regardless of whether you are using HITL or SITL.  I can't seem to get any hangs on windows clicking play/stop over and over, but I'll review the code more closely and see if I can find anything.  Have you been able to debug where it is hanging exactly (stack trace?)

bwassim, 2020-09-02:
I am not exactly sure how to proceed to actually debug, would appreciate some help here. Thanks