microsoft/AirSim #2406: build.cmd Issue with Unreal 4.24.2 and vs2019

InfraredLaser, 2020-02-06:
I am getting errors upon first building AirSim for unreal 4.24.2, vs2019, and using windows 10.
Here are the error messages:

[image]

To build AirSim I am using the _x64 Native Tools Command Prompt for VS 2019_.
I have tried to fix the error in the file ThreadUtils.hpp with an `#include <string>` but in doing so the error list grew to 55 errors.

I followed the steps to build AirSim on windows though this guide: https://microsoft.github.io/AirSim/docs/build_windows/

Any ideas on how to get a successful build?

rajat2004, 2020-02-06:
Maybe try with https://github.com/microsoft/AirSim/pull/2402? Could reduce the number of errors

InfraredLaser, 2020-02-06:
Hi @rajat2004 ,

I tried #2402  just now and I still get the same error as above. I believe the issue is with this error: 
_error C2039: 'string': is not a member of 'std' [D:\AirSim\MavLinkCom\MavLinkCom.vcxproj]_

When I open the file here is what I find:
[image]

It is this file that when I add `#include <string>` at the specified location 55 errors occur after performing build.cmd

rajat2004, 2020-02-07:
So the `string` error goes away after including the header right?
Maybe post the text of the errors beign produced after fixing this, and then could try to figure out the problems

InfraredLaser, 2020-02-07:
That's correct @rajat2004 , the error is within the file _ThreadUtils.hpp_ and when I added `#include <string>` the `name` variable in the red underline will then be recognized as a `string`. However, upon building the code using the _x64 Native Tools Command Prompt for VS 2019_ I get the 55 errors. 

The 55 errors summarize to being linker and thread issues (unfortunately I don't have a photo as I have already uninstalled VS2019 *reason below*), and I believe the source of the error is that the files in AirSim are being built using v141 of visual studio 2017 instead of using v142 of visual studio 2019. Here is my steps that I took to come to this conclusion: 

1. When I build the files using the command line instead of _x64 Native Tools Command Prompt for VS 2019_ the print message reads "the files are built using Visual Studio 2017". 
2. I then installed the build tools for VS 2017 and ran the build.cmd  in the AirSim directory on _x64 Native Tools Command Prompt for VS 2017_
3. The print message then reads that the files were built with v141 of Visual Studio, while I have installed v142 for Visual Studio 2019. 

I think my solution is to download VS 2017 with v141 and build AirSim on _x64 Native Tools Command Prompt for VS 2017_ as I don't see any solutions for AirSim to work with VS 2019.

rajat2004, 2020-02-08:
Yeah, the problems should be gone if using VS 2017, though things can't be fixed to work with VS 2019 if it's not tried out first :)

InfraredLaser, 2020-02-11:
After installing vs2017 the problem does go away. I think AirSim is not ready for vs 2019.

thejeshk, 2020-02-21:
Hi @madratman, 

I am planning to use AIRSIM for one of my project. I am using Windows VS2017 . Could you please confirm can I use Unreal 4.22 instead of 4.18 ? Document says 4.18 

Thanks and regards, 
Thejesh Kumar

InfraredLaser, 2020-02-21:
Hi @thejeshk,

You can most definitely use the latest version of unreal engine using visual studio 2017. Follow the directions in this thread: #2402

closed 2020-08-11 by madratman