microsoft/AirSim #2407: Linux v1.2.0 City binaries corrupted

d-rohr, 2020-02-06:
## The Issue
I am trying to run the City binary on 18.04 Ubuntu. When I try to unzip the provided files, I get an error. I have seen on other threads that the binary is corrupted, but to my knowledge this has not been reported as a bug.

## Steps to reproduce
*sudo apt install -y wget unzip
*wget -c https://github.com/Microsoft/AirSim/releases/download/v1.2.0Linux/CityEnviron.zip.001
*wget -c https://github.com/Microsoft/AirSim/releases/download/v1.2.0Linux/CityEnviron.zip.001
*cat CityEnviron.zip.00* > CityEnviron.zip
*unzip CityEnviron.zip

## Terminal output
    $ unzip CityEnviron.zip    
    Archive:  CityEnviron.zip    
      End-of-central-directory signature not found.  Either this file is not    
      a zipfile, or it constitutes one disk of a multi-part archive.  In the    
      latter case the central directory and zipfile comment will be found on    
      the last disk(s) of this archive.    
    unzip:  cannot find zipfile directory in one of CityEnviron.zip or    
            CityEnviron.zip.zip, and cannot find CityEnviron.zip.ZIP, period.

d-rohr, 2020-02-06:
[Here](https://github.com/microsoft/AirSim/issues/1668#issuecomment-467134766) is a link to unzipping failing with 7z as well.

dimikout3, 2020-04-26:
@madratman  @ironclownfish  We are conducting research on AirSim. Are you planing to release CityEnviron binaries for Linux(Ubuntu) soon ?

Tomakko, 2021-09-13:
Any updates on getting the CityEnvion working under Linux?