microsoft/AirSim #2408: Controlling the drone in AirSim using APIs

charbelmansour, 2020-02-06:
Hello,
We built AirSim on windows, we used LandscapeMountains and when we run the code in VisualStudio 2017, we obtain the environment and the drone that we want to control.
We are facing a problem in controlling the drone using APIs. We followed the instructions and we added the folder of HelloDrone in the plugins inside the source folder. After that we added the following code in the main of HelloDrone.

#include <iostream>
#include "vehicles/multirotor/api/MultirotorRpcLibClient.hpp"

int main()
{
	using namespace std;
	msr::airlib::MultirotorRpcLibClient client;

	cout << "Press Enter to enable API control" << endl; cin.get();
	client.enableApiControl(true);

	cout << "Press Enter to arm the drone" << endl; cin.get();
	client.armDisarm(true);

	cout << "Press Enter to takeoff" << endl; cin.get();
	client.takeoffAsync(5)->waitOnLastTask();

	cout << "Press Enter to move 5 meters in x direction with 1 m/s velocity" << endl; cin.get();
	auto position = client.getMultirotorState().getPosition(); // from current location
	client.moveToPositionAsync(position.x() + 5, position.y(), position.z(), 1)->waitOnLastTask();

	cout << "Press Enter to land" << endl; cin.get();
	client.landAsync()->waitOnLastTask();

	return 0;
}

This code is available in this site: https://microsoft.github.io/AirSim/docs/apis_cpp/#using-c-apis-for-airsim
and it is mentioned that it should be added in the main of HelloDrone.
We would like to know why it is not working or if there exists another method to let the drone fly using the APIs.
Thank you for your help.
Best Regards,

rajat2004, 2020-02-07:
Using the C++ APIs will need a bit of experience and patience. If using Linux, understanding CMake will also be required
Easier way is to use the Python APIs

jonyMarino, 2020-09-24:
Hi @charbelmansour. I think you confused something by saying "we added the folder of HelloDrone in the plugins inside the source folder". In the plugins folder you just need the AirSim plugin. And after running LandscapeMountains you can run your c++ code. I just tried to run your code modifying the main.cpp in the HelloDrone project without any problem. If you are still facing an issue please feel free to reopen this.

closed 2020-09-24 by jonyMarino