microsoft/AirSim #2409: How to use

xiaoaoran, 2020-02-07:
I tried to get segmentation ground truth via `client.simGetLidarSegmentation()`:

`lidarData = self.client.getLidarData()`
`lidarSeg = self.client.simGetLidarSegmentation()`

I can get `lidarData` but errors occured in `lidarSeg`  as:
> Traceback (most recent call last):
  File "car_lidar_manctrl.py", line 80, in <module>
    lidarTest.execute()
  File "car_lidar_manctrl.py", line 36, in execute
    lidarSeg = self.client.simGetLidarSegmentation()
  File "D:\Aoran\AirSim\PythonClient\airsim\client.py", line 166, in simGetLidarSegmentation
    return self.client.call('simGetLidarSegmentation', lidar_name, vehicle_name)
  File "C:\Program Files\Python37\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Program Files\Python37\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: server could not find function 'simGetLidarSegmentation' with argument count 2.

Here is my setting:
> {
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "",
  "DefaultSensors": {
    "Barometer": {
      "SensorType": 1,
      "Enabled": true
    },
    "Gps": {
      "SensorType": 1,
      "Enabled": true
    },
    "Lidar1": {
      "SensorType": 6,
      "Enabled": true,
      "NumberOfChannels": 16,
      "PointsPerSecond": 10000,
      "DrawDebugPoints": false,
      "DataFrame": "SensorLocalFrame"
    }
  },
  "SubWindows": [
    {
      "WindowID": 0,
      "CameraName": "0",
      "ImageType": 3,
      "Visible": false
    },
    {
      "WindowID": 1,
      "CameraName": "0",
      "ImageType": 5,
      "Visible": false
    },
    {
        "WindowID": 2,
        "CameraName": "0",
        "ImageType": 0,
        "Visible": false
    }
  ]
}

There are no instructions about this api. Anyone could tell me? Thanks!

xiaoaoran, 2020-02-09:
Please help me with this problem

xiaoaoran, 2020-02-13:
Hello?

closed 2020-02-21 by xiaoaoran

TheFloe1995, 2020-05-11:
I also stumbled across this issue. It seems that you have to call both functions _right after each other_. 

In a quick experiment I called both functions directly after each other at 10 FPS while moving. Most of the time the number of points was equal in both lists. But in 7/300 cases they were different. 

Seems to be timing dependent which is quite inconvenient in my opinion. Are there any plans to change that behaviour, e.g. having a single function call to get the point cloud + segmentation IDs?

rajat2004, 2020-06-29:
I've opened #2810 which combines segmentation data into `LidarData` struct fetched using `getLidarData`