microsoft/AirSim #2412: Unreal SUV is Shaking

ITali21, 2020-02-08:
I'm use unreal engine 4.18 and airsim 1.2.0 versions. When I start driving SUV, the SUV and the camera start to shaking extremely. I didn't change the SUV settings. I don't found problem reason.

Video of problem:
https://drive.google.com/file/d/1IEeAgwM2I3kzpKS2-FRIo6CTARADADQV/view?usp=sharing

Anyone could tell me why? and can i solve this problem. Thanks!

ironclownfish, 2020-02-25:
I'm guessing this has to do with the collision detection with the ground. Does your ground have bumps or small holes in it? Can you try putting one of Unreal's default plane objects in the scene and driving on it, and see if the issue occurs in that case? (The default plane is definitely perfectly flat)

jonyMarino, 2020-08-13:
closed due to lack of response

closed 2020-08-13 by jonyMarino