microsoft/AirSim #2413: Connection between AirSim and PX4 over the network.

xogns8121, 2020-02-11:
airsim = latest version 1.2.x / window 10
Unreal = v4.18
px4 = latest version 1.10.x / ubuntu 16.04

I did a lot of things to connect between airsim on window 10 and px4 on ubuntu 16.04 over network. Before, I tried to connect gazebo on ubuntu 16.04 and px4 on ubuntu 16.04 over the network and it work properly. I checked px4 - gazebo over network work well. So i think there is problem at AirSim or I'm making a mistake. but I have no idea what i'm doing wrong. please i need help..

<< gazebo >>

I changed gazebo iris.sdf file to connect remote pc.
`<mavlink_addr>203.255.57.83</mavlink_addr>`

[image]

[image]

<< px4 >>

[image]

Gazebo - px4 works well but airsim doesn't..

-----------------------------------------------------------------------------------------

1. changed airsim setting.json 

1-1
I changed airsim setting.json but it does not work so i even tried changing AirSimSettings.hpp file also. but still it doesn't work. 

`{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "UseTcp": false,
      "UdpIp": "203.255.57.83",
      "UdpPort": 14560
    }
  }`
This is my airsim setting.json file. 
I found that UdpIp and UdpPort value are for connection with px4 sitl.

sometimes Airsim 
First, **I suspect that the program is making a self connection.**

1-2
I even tried changing defualt values about mavlinkconnection.
I changes AirSimSetting.cpp file.

[image]

here i Changed default value
Ex) udp_address = "203.255.57.xx"

1-3
I checked values visual studio debug tool.

[image]

[image]

second, **What should i change or What should i do to connect??**

2. px4 
I type this code
`Simulator start -u 14560`
Then px4 is waiting for connection from airsim.

[image]

rajat2004, 2020-02-13:
I did have a brief try at running PX4 with AirSim, some thoughts on this matter-

> I changed gazebo iris.sdf file to connect remote pc.
<mavlink_addr>203.255.57.83</mavlink_addr>

Is this the IP address of the machine running Gazebo? If yes, then probably something like this will be required for AirSim as well

Seems like PX4 from 1.9.x uses TCP for LockStep scheduling, and the method of connecting using UDP doesn't work right now, even, on the same machine from what I tried, maybe it's the case that I got my ports wrong will have to look more into it

For the TCP connection, AirSim listens for connections on the TCP Port, from the code atleast, so it seems that somehow we need to specify the address of the machine running the simulator (AirSim) while starting the PX4 SITL using `make` 
Or fix the method of UDP connection currently being used in AirSim

The UDP method is the same from the previous versions of PX4 for backwards compatibility, so I suspect if this is the case and it needs fixing, then that would break, which I don't think would be a very bad thing, since that PX4 version is also pretty old
But as it stands currently, the older versions of PX4 should also work, so maybe have a try with 1.8.2 or something?
Also disable the Windows Firewall if not already done

@lovettchris Any idea regarding this?

xogns8121, 2020-02-13:
Thank you fore reply 

> Is this the IP address of the machine running Gazebo? If yes, then probably something like this will be required for AirSim as well
Yes, this ip is gazebo and i also changed 
`<use_tcp>false</use_tcp>`
i also thought that i have to set airsim just link changed gazebo.

I tried not only tcp connection but also udp connection.
In cast tcp connection, airsim linsten that you mentioned and px4 try to connecting.
**Contrary, in case udp connection, i found px4 listen udp and airsim have to trying to connecting.**
at this moment, 
in px4, i typed
`simulater start -u 14560`
in airsim ,I thought I have to change airsim option 
`"UsingTcp": false,
"UdpIp": "203.255.57.xx"
"UdpPort": 14560"`
 px4 waiting udp well, but airsim get error that i showed above.

why i don't use tcp is that
px4 support mavlink(udp) connection over network
but doesn't support simulator(tcp) connection over network.

i wonder airsim support only tcp for simulator connection??

rajat2004, 2020-02-14:
AirSim does support UDP, but I think it'll need to be fixed for newer versions of PX4. But it is quite strange that PX4 doesn't support TCP over the network. From the brief look I had over the PX4 and AirSim code, not sure why UDP isn't working, that'll need more in-depth investigation

But older versions of PX4 should still work with AirSim, the instructions and settings for which can be found by going through the history of the `px4_sitl` Readme. Might need to be modified to work with the new code, or if you don't want to do that, maybe could just checkout the repo to the commit before support for PX4 Lockstep was added.

You probably have a specific need for PX4 since you have been looking into it for so much time, otherwise could have a look at using SimpleFlight or even ArduPilot if they can do what is required

xogns8121, 2020-02-17:
There's need to do with old version.
thank you for sincere.

jonyMarino, 2021-06-03:
This was fixed in [the latest version (1.5.0)](https://github.com/microsoft/AirSim/releases)

closed 2021-06-03 by jonyMarino