microsoft/AirSim #2414: Missing pom.xml

charbelmansour, 2020-02-11:
Hello,
We followed these steps:
1) git clone https://github.com/alexandrelombard/airsim-jvm-api

2) cd airsim-jvm-api

3) mvn clean install

When we reached step 3 we obtained the following error:

[image]

Can you please give us advice about this issue.
The pom.xml is missing from the project.
Thank you for your help
Best Regards,

rajat2004, 2020-02-12:
Since this is a problem with the other repository and not the main AirSim one, maybe open an issue there?

charbelmansour, 2020-02-16:
Yes, thank you. We opened an issue in the other repository

closed 2020-02-26 by charbelmansour