microsoft/AirSim #2417: Build failed on Ubuntu 18.04

SalimTerryLi, 2020-02-14:
Following [this](https://microsoft.github.io/AirSim/docs/build_linux/) instruction.
UE was build successfully but in the AirSim section, `./build.sh` doesn't work.
```
************************************
AirSim setup completed successfully!
************************************
```
Above was printed once `./setup.sh` was done, and no identical error reported, but `./build.sh` dumped and exited:
```
+ MIN_GCC_VERSION=6.0.0
+ gccBuild=false
+ [[ 0 -gt 0 ]]
+ '[' '!' -d ./external/rpclib/rpclib-2.2.1 ']'
+ '[' -d ./cmake_build ']'
++ uname
+ '[' Linux == Darwin ']'
++ readlink -f cmake_build/bin/cmake
+ CMAKE=/media/salimterryli/dev-linux/AirSim/cmake_build/bin/cmake
+ false
+ [[ ! -d llvm-source-50 ]]
+ [[ -d ./llvm-build/output/lib ]]
+ echo 'ERROR: clang++ and libc++ is necessary to compile AirSim and run it in Unreal engine'
ERROR: clang++ and libc++ is necessary to compile AirSim and run it in Unreal engine
+ echo 'Please run setup.sh first.'
Please run setup.sh first.
+ exit 1
```
"llvm-build" is an empty folder and "llvm-source-50" only contains one directory named "projects"

I have configured different versions of clang with `update-alternatives` and now clang-5.0 is selected.
```
salimterryli@STL-Laptop:~$ clang -v
clang version 5.0.1-4 (tags/RELEASE_501/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
Found candidate GCC installation: /usr/bin/../lib/gcc/i686-linux-gnu/8
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/7
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/8
Found candidate GCC installation: /usr/lib/gcc/i686-linux-gnu/8
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/7
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/7.4.0
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/8
Selected GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/7.4.0
Candidate multilib: .;@m64
Candidate multilib: 32;@m32
Candidate multilib: x32;@mx32
Selected multilib: .;@m64
Found CUDA installation: /usr/local/cuda, version unknown

```

SalimTerryLi, 2020-02-14:
I have tried `./build --gcc` and it works.

SalimTerryLi, 2020-02-14:
Then it crashed when I clicked Play button:
``
/media/salimterryli/dev-linux/UnrealEngine/Engine/Binaries/Linux/UE4Editor: symbol lookup error: /media/salimterryli/dev-linux/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so: undefined symbol: _ZN3rpc6detail10dispatcher19enforce_unique_nameERKNSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEE
```

rajat2004, 2020-02-14:
What's the output when you run `setup.sh`?
If ti's running into problems, maybe try using `clean_rebuild.sh`

SalimTerryLi, 2020-02-14:
> What's the output when you run `setup.sh`?
> If ti's running into problems, maybe try using `clean_rebuild.sh`

I use proxychains for higher transforming speed. [This](https://paste.ubuntu.com/p/qdnpsC4xQx/) is the full output from setup.sh.
Then I will not able to build as following dumped:
```
salimterryli@STL-Laptop:/media/salimterryli/dev-linux/AirSim$ ./build.sh 
+ MIN_GCC_VERSION=6.0.0
+ gccBuild=false
+ [[ 0 -gt 0 ]]
+ '[' '!' -d ./external/rpclib/rpclib-2.2.1 ']'
+ '[' -d ./cmake_build ']'
++ uname
+ '[' Linux == Darwin ']'
++ readlink -f cmake_build/bin/cmake
+ CMAKE=/media/salimterryli/dev-linux/AirSim/cmake_build/bin/cmake
+ false
+ [[ ! -d llvm-source-50 ]]
+ [[ -d ./llvm-build/output/lib ]]
+ echo 'ERROR: clang++ and libc++ is necessary to compile AirSim and run it in Unreal engine'
ERROR: clang++ and libc++ is necessary to compile AirSim and run it in Unreal engine
+ echo 'Please run setup.sh first.'
Please run setup.sh first.
+ exit 1
```

I have launched `clean_rebuild.sh` and it also report the same problem. It seems `clean_rebuild.sh` combines the cleanup, setup and build to a single file.

SalimTerryLi, 2020-02-14:
Thank you @rajat2004 , it's my fault that proxychains is blocking llvm-sources-50 being cloned.

closed 2020-02-14 by SalimTerryLi